portapack-mayhem/mayhem-firmware #661: offline user manual as HTML in >UTILITIES

Watchkido, 2022-05-27:
Paste the WIKI into the portapack as an HTML page.
- Format HTML on the SD card
-language: All / German (I'm trying to translate into German)
-In the title bar, a quick switch option to the manual
-Contents: user manual, precise instructions, links to explanatory videos, list of frequency ranges for scanners, links to community groups, links to sample downloads
-Kept updated by [github-wikito-converter](https://github.com/yakivmospan/github-wikito-converter)

ArjanOnwezen, 2022-05-27:
I don't see the point of this. Why keep it offline as HTML, while everything can be kept online?

ArjanOnwezen, 2022-05-27:
Just check github wiki for the latest version. I do see it might be handy to have something static like Github Pages with Jekyll or so, but at least it's online with the latest version

ArjanOnwezen, 2022-05-27:
Else many people will ask questions based on outdated manuals..

eried, 2022-05-27:
@Watchkido where are you translating the manual? I can see the point of an offline manual for use offline, on the field, etc but @ArjanOnwezen has a point, for now the manual is a bit incomplete too

Watchkido, 2022-05-28:
A user manual is always for beginners.
As a radio amateur, I know the background of every Portapack app, but not how to use it.
That's why I expect good software to be self-explanatory or at least to have a good explanation. As a "beginner", I don't want to try out new software with a smartphone in one hand and a hackrf in the other. That's why I think the operating instructions should be in the device.
Since the user manual for each version is up to date, it should not be a problem that undocumented apps or functions are included.
However, the question: "where are you translating the manual?" is a killer question.
I don't have an answer to that.

But I have a better suggestion: please build in a text editor. That way you can change TXT. files in the device and I can load my own translation into the device. 
My German is much better than my English :-)

ArjanOnwezen, 2022-05-28:
If you want a manual create on in Github in Markdown language. Use DE: prefix for the German version. Markdown can be easily converted to other formats, like HTML and PDF.

eried, 2022-05-28:
> However, the question: "where are you translating the manual?" is a killer question.

You say you are trying to translate it to German, where? In the wiki? In a fork? In your own offline files?

ArjanOnwezen, 2022-05-28:
Just create a manual in this repo, in subfolder manual?. And prefix all German pages with DE:, all Dutch pages with NL: etc..

Watchkido, 2022-05-28:
@eried say: "You say you are trying to translate it to German, where? In the wiki? In a fork? In your own offline files?"

I don't know.
You know your way around here better.
Find a place where it is comfortable for you and easy for me. 
Wiki, fork, extra page, whatever
Best for me:
1. download in DeepL, 
2. look over and correct
3. upload to WIKI
4. finished

Men! Don't talk so much. 
Just do it.
Or reject it.
you are the boss here
Time is precious.
Your software has so much potential.
I love it.

gerryk, 2022-06-09:
_Just do it.
Or reject it.
you are the boss here_

Not a dictatorship. All issues should have their merit decided by consensus. If there is something you want to see done, persuade people as to the merit of your idea, propose a bounty to reward someone for implementing it, or do it yourself.

gullradriel, 2023-05-10:
That talk have not move an inch for months. Consensus reached: I am closing it :-)

closed 2023-05-10 by gullradriel