portapack-mayhem/mayhem-firmware #663: Display not showing Portapack H2

stenbus, 2022-05-28:
Hi! 
I've been running the 1.4.2 version since I got my hackrf, and it worked seamlessly. But now when I try to update to this version the screen only blinks. I can still navigate the many blindfolded but no display. 
I've tried multiple things like DFU mode to normal hackrf firmware. that works. downgrading to the stable 1.4.2 works as well. 
Ive tried some nightly releases with no luck. What could this be?

one more thing. When I download the newest releases the "msvcp120" and "msvcr120" are missing, so I've used the old ones from later releases. is that ok or?

jLynx, 2022-05-28:
Closing as this is an invalid issue (not an issue) and you didn't read the wiki before creating this https://github.com/eried/portapack-mayhem/wiki/Won't-boot

closed 2022-05-28 by jLynx

jLynx, 2022-06-10:
It's possible that you had corrupt memory. Removing the rtc battery resets that