portapack-mayhem/mayhem-firmware #668: Record button not working from within the looking glass

sherlock-ohm, 2022-05-31:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly version here: https://github.com/eried/portapack-mayhem/releases)

----

**Describe the bug**

Record button not working from within the looking glass. I have tried with a H2 on Mayhem firmware v1.5.3 and Nightly Release - 2022-05-29.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to looking glass.
2. Adjust the marker to a frequency that you'd like to record. 
3. Click the middle button which takes you to the SPEC screen.
4. Go to record and press the middle button.

**Expected behavior**
I would expect the device to record the signal and place it on the SD Card in the root directory.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release: Mayhem firmware v1.5.3
* Latest Nightly release: 2022-05-29
* Previous working release: Unknown

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.

tony1492, 2022-06-03:
If your are in SPEC Screen there is no audio Playing in the first place you must go to WFM,  NFM, AM, DSB,USB, LSB or CW to record.  Hope this helps.

Tony

AWH2332, 2022-08-03:
I have a similar incident when i´m trying to record some NRF packages in v-1.5.4

    Go to recibe > NRF.
    Wait to the hack rf displais some info 
    Go to record and press the middle button.

gullradriel, 2023-03-31:
It's not a bug, it's a feature. Audio is not recording in spec. Spec is somewhat the ancestor of Looking glass.
As the others referenced bugs about BTLE or NRF, they already have their own issues.

closed 2023-03-31 by gullradriel