portapack-mayhem/mayhem-firmware #669: Add 10Hz and 1Hz steps. This would be useful for working QO-100 satellite.

devop-mike, 2022-06-05:
Add 10Hz and 1Hz steps.

This would be useful for working QO-100 satellite.

Hopefully this is the only file that needs changing.
https://github.com/eried/portapack-mayhem/blob/39134b62c365f98b74119bd247d7ab9434072a1c/firmware/application/ui/ui_receiver.hpp#L250

[image]

Brumi-2021, 2022-08-27:
Hi , @devop-mike , I did nit check yet the code if is only that common file . Maybe it is a simple change as you said , but do you need for both TX and RX or just for TX ?

Brumi-2021, 2022-08-30:
Hi , 
Did you compile your suggestion ???
@devop-mike , your suggestion is correct , just applying your change , 
we really  could have those steps in all RX ,and also is shared to all TX , 

but two comments about it , 

1-) Do you really need 1 Hz step ?  (while probably we will have easily  a thermal drift much higher than that ....) 
2-) Currently the freq. digit space is limitted to XXXX. ABCD  Mhz  ,  therefore  the last 4 digits shows, 
A -->A  x 100khz 
B ..> B x  10 khz 
C- -->C x  1 khz
D  --> D x   100 Hz  

After applying your suggestion ,  we will miss 2 decimal digits more , to show the real  changed freq. in the display and we do not have that display  space ...   
[image]

Then I suggest maybe go down to 50 (you will see display change every two steps with current decimals ) 
 , but not  10 and 1 .... How do you feel about it ? 
Pls. check 
[portapack-h1_h2-mayhem_QO-100.zip](https://github.com/eried/portapack-mayhem/files/9456052/portapack-h1_h2-mayhem_QO-100.zip)

Cheers

devop-mike, 2022-08-31:
I complied the firmware and had it tested. 1 Hz was too fine, but 5 Hz was useful.  So I would pick 5, 10, 50, 100.  In practice the SSB fine tuning is done by ear.  It's nice to be able to get the voice spot on.  None of my physical radio equipment shows the step once selected, so that's a luxury I didn't realise I was missing.

Thermal drift is not an issue, every thing is getting locked to a GPS 10 MHz reference.  Surprisingly Doppler shift is more of an issue working a geostationary satellite.  Not as fixed a distance as I had assumed.

Since I can compile the firmware here, I can add in the steps I need.  So don't waste too much time on this.  The dream would be a config screen where you can turn on or off the steps you want.

I see your display pains, and it gets worse with the really high frequencies for example you would need something like 10G489M703k816 Hz.  QO-100 workable range is 10.4895 to 10.49 GHz.

Brumi-2021, 2022-08-31:
Hi @devop-mike , 
Thanks for your answer.
I will recheck on next week end about its practical hear usage in HF SSB 80m , 40m , 20m with an up converter.  Last time that I tested I missed some more fine step  tuning, but just a little.  

I think that one compromise solution  might be to submit a PR extending the steps to the minimum necessary, 50,10, 5 Hz ???  But do not touch any GUI display digits.
Then it will be hidden for the normal usage , but you will still have it  . Let me also recheck and come back to you . 
73,

Brumi-2021, 2022-09-03:
Hi @devop-mike , I checked the voice pitch adjustment to demodulate in Audio App ,  LSB in HF 80m , (3.776 Mhz)  
and really the current Local Oscillator  frequency steps of the min one ,  100 Hz is too much .

 I tried also 50 , 10 , 1 Hz ,  and too me 10 is enough . (I thing 5 Hz step might  be too small one , but it is just my opinion, not tried ) .   Then I agree , to add your modification , without touching the GUI displayed digits .  

Then , will you release the PR ?  (I think it will be  a minor change, no risk , and clear small benefit to all) .

 (let me attach recorded audio example . I was moving steps 10 Hz while the ham operator  was talking , and it is easy too adjust his  natural voice .(and the difference is already low noticeable freq. shift )  
  
Cheers, 
[AUD_20220904T233549_ 203776739Hz.zip](https://github.com/eried/portapack-mayhem/files/9483525/AUD_20220904T233549_.203776739Hz.zip)

closed 2022-09-12 by eried