portapack-mayhem/mayhem-firmware #674: crashes when transmitting with HackRF mode with v1.5.4

jamesliu96, 2022-06-08:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly verison here: https://github.com/eried/portapack-mayhem/releases)

----

**Describe the bug**
crashes when transmit signal in HackRF mode with hackrf_transfer or GNU Radio

**To Reproduce**
Steps to reproduce the behavior:
1. HackRF -> YES
2. `hackrf_info`
```
hackrf_info version: unknown
libhackrf version: unknown (0.6)
Found HackRF
Index: 0
Serial number: 0000000000000000f77c60dc295a51c3
Board ID Number: 2 (HackRF One)
Firmware Version: v1.5.4 (API:1.06)
Part ID Number: 0xa000cb3c 0x006a4f5e
Operacake found, address: 0xff
Operacake found, address: 0xff
Operacake found, address: 0xff
Operacake found, address: 0xff
Operacake found, address: 0xff
Operacake found, address: 0xff
Operacake found, address: 0xff
Operacake found, address: 0xff
```
3. `hackrf_transfer -t /dev/zero` or launch a GRC app
```
call hackrf_set_sample_rate(10000000 Hz/10.000 MHz)
call hackrf_set_hw_sync_mode(0)
call hackrf_set_freq(900000000 Hz/900.000 MHz)
Stop with Ctrl-C
19.9 MiB / 1.005 sec = 19.8 MiB/second
Exiting...
Total time: 5.02471 s
hackrf_stop_tx() done
hackrf_close() done
hackrf_exit() done
fclose() done
exit
```
4. screen flashes and turns white, radio transmission stops

**Expected behavior**
transmits normally

As I revert the firmware back to v1.5.3, everything becomes normal.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release: v1.5.4
* Latest Nightly release: current
* Previous working release:

jLynx, 2022-06-08:
If you try 1.5.3, does that solve the issue?

jamesliu96, 2022-06-09:
@jLynx Yes, it does. v1.5.3 doesn't reproduce the problem.

jLynx, 2022-06-09:
I am wondering if it's the HackRF core that we have updated that has caused the issue. In that case maybe the issue should be created at https://github.com/greatscottgadgets/hackrf

Would it be possible for you to temporarily flash the HackRF only firmware linked above without the portapack and see if it works?

ArjanOnwezen, 2022-06-10:
> I am wondering if it's the HackRF core that we have updated that has caused the issue. In that case maybe the issue should be created at https://github.com/greatscottgadgets/hackrf
>
> Would it be [...]

Yeah that's very well possible. I guess a roll back is needed and stick to official hackrf releases.
Since hackrf updates also can break normal hackrf operations.

jLynx, 2022-06-10:
I'll try to remember to create a PR to roll back the HackRF submodule

gullradriel, 2023-04-26:
Version has been replaced by 1.6.0 and users are not reporting the problem anymore. 
BTW hackrf submodule is up to date in the nightly for a while, and nothing to see so far. Maybe we will see a few things coming from latest hackrf board users, but yet to come.

closed 2023-04-26 by gullradriel