portapack-mayhem/mayhem-firmware #675: Pocsag in WFM-mode??

loewal, 2022-06-09:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly verison here: https://github.com/eried/portapack-mayhem/releases)

----

**Describe the bug**
There is no "Pocsag-sound": bleeps...only some interruption in the white noise as if the modulation is wrong.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Receive
2. Tap on Pocsag
3. etc.

**Expected behavior**
Clear bleeps and decoded messages.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release: no luck
* Latest Nightly release: no luck
* Previous working release: went back to 1.4.3 and 1.4.1: same

jLynx, 2022-06-09:
Is this different to your other issue you have open?

loewal, 2022-06-09:
This is what I receive with SDRangel with the same Hackrf, antenna, etc.
[image]

loewal, 2022-06-09:
No, I just want to report, that its hasn't been solved yet....

jLynx, 2022-06-09:
Please keep all the info in the original issue. No need to keep opening them as the original one is not closed. So I will go ahead and close this duplicate

closed 2022-06-09 by jLynx