portapack-mayhem/mayhem-firmware #676: POCSAG RX not decoding messages after version 1.4.1

zigad, 2022-06-10:
**Describe the bug**
Since refactoring of POCSAG RX App in 1.4.2 (#426) POCSAG messages are not decoded, only 1% of messages is detected and 0 messages are 100% correctly decoded.

Whenever I revert back to 1.4.1 I never have problems decoding POCSAG messages.

This issue has been discussed multiple times (#675, #667, #478)

I think the main problem here is that Portapack doesn't have enough processing power to decide what kind of POCSAG message it's receiving, what kind of polarisation and what baud rate it should decode.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to a place where they transmit POCSAG messages (burger restaurants/McDonalds, fast food restaurants and so on)
2. Find correct frequency and open POCSAG RX, wait for message and see if it was decoded on any version later than 1.4.1,
3. revert back to 1.4.1 and test it out.

**Expected behaviour**
It's expected for POCSAG RX app to decode all of messages, for me numeric/tone only messages are never decoded, alphanumeric messages are in really rare cases are detected and decoded with gibberish even RIC/CAP codes are incorrect.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release: not working
* Latest Nightly release: not working
* Previous working release: 1.4.1

**Additional**

Attaching videos to different version.
Please note videos were taken the same day, exactly the same location from transmitter, same settings and same antenna was used on all versions!

[Version 1.5.4 **_Not working_**](https://youtube.com/shorts/otX-rH5Q794)

[Version 25a7eeac **_Not working_**](https://youtube.com/shorts/dT0l39FZjwM)

[Version 1.5.3 **_Not Working_**](https://youtube.com/shorts/QF-A6PGCJWk)

[Version 1.4.1 **_WORKING_**](https://youtube.com/shorts/qSDwq2hgMhM)

Please also note that all versions above 1.4.1 are not decoding. I just don't have enough time to record it.

This is also happening when trying to decode Slovenian Government pagers for national emergency and fire department, so it's not just restaurant/hospital pagers.

**Proposed Solution**
Revert back POCSAG RX decode logic as it was before, we can write a wiki page on instructions for POCSAG app so users know what means 1200, P and so on. Right now POCSAG RX app is useless since 1.4.1 when it actually worked.

TQMatvey, 2022-06-20:
i think we need to revert all commits from https://github.com/eried/portapack-mayhem/pull/426
this will fix issue at least...

GullCode, 2022-06-20:
I think it's more coming from the portapack.cpp modifications than that PR.
I noticed and told people that on latest we have the white noise problem back in all demodulations, and that is causing bad or no decoding.
To verify what I am saying, can you please try the latest bin in Recon app channel / Test drive ?
I'll upload it one more time in a few hours and post a direct link here.
IIRC the #426 is applied but not the latest portapack.cpp modifications, so your test will allow us to precise what is the bad PR here

gullradriel, 2023-04-26:
Users are still having the problem. @zigad and @loewal, are you up for testing ? Can @heurist1 be joined ?

loewal, 2023-04-26:
Yes. By the way: Pocsag is working, but sometimes it takes maybe an hour and then something is triggered to keep on going….

gullradriel, 2023-04-26:
I think it's really having to do with what's said in the beginning. Baud and polarity autodetection is not working that well due to poor power processing.
Maybe we can make a mix and allow users to choose between actual mode and manual old mode

nexpspace, 2023-04-28:
> I think it's really having to do with what's said in the beginning. Baud and polarity autodetection is not working that well due to poor power processing. Maybe we can make a mix and allow users to choose between actual mode and manual old mode

The current autodetection works so far flawless for me, so I would prefer that the current autodetection mode will be available too — so thumbs up for a switch between auto mode and manual mode.

zigad, 2023-04-28:
Aggree with that
We can create a button called "advanced mode" where users have to select baud rate and polarity. Otherwise it's in auto mode

> The current autodetection works so far flawless for me, so I would prefer that the current autodetection mode will be available too — so thumbs up for a switch between auto mode and manual mode.

gullradriel, 2023-07-18:
Duplicated of https://github.com/eried/portapack-mayhem/issues/667

closed 2023-07-18 by gullradriel