portapack-mayhem/mayhem-firmware #679: Problems with the map in ADSB

darknebular, 2022-07-08:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly verison here: https://github.com/eried/portapack-mayhem/releases)

----

**Describe the bug**
Problems with the map in ADSB. The world_map.bin doesn´t load

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Receive'
2. Tap on 'ADSB
3.' Click on 'Show in MAP'

**Expected behavior**
A map, not a blur og blue-grey image.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release: 1.5.4
* Latest Nightly release: N/A
* Previous working release: N/A

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.

eried, 2022-07-08:
Has the map worked before in your unit?

darknebular, 2022-07-08:
> Has the map worked before in your unit?

I am new with portapack and hackrf.
I only tested 1.5.4

darknebular, 2022-07-08:
> Has the map worked before in your unit?

I proof others maps, one of around 455 MB with a .jpg of around 23,8 MB and yours 2GB and with a .jpg of around 98MB.

I only see a red triangle in a blur blue-grey image, the map doesn't load well.

eried, 2022-07-08:
But has any of the map work before? Post a picture

darknebular, 2022-07-09:
> But has any of the map work before? Post a picture

Never work before.

darknebular, 2022-07-17:
Here I send to you an image of the error.
[image]

darknebular, 2022-07-21:
Please reopen this ticket.

This is an other problem.

jLynx, 2022-07-21:
I think your have a bad unit or bad SD card as you are the only person with this issue (and same with the others). I think if you fix the route cause, then it will fix all your issues. Keeping it closed as it's an isolated issue

darknebular, 2022-07-22:
I have two portapacks, the problem with both are same with your firmware.
Perhaps, new H1?

BehleZebub, 2022-10-22:
I think the problem is, how python3.write writes binary integer.
Python3 does take this setting from sys.byteorder that's why on some machines it will work on others it won't.
The mayhem app expects the numbers to be unsinged short little endian.

So if you change every struct.pack('H'...) to struct.pack('<H'...) it will work on all systems.

Also outfile is not closed which should be done at the end.

[generate_world_map.bin.txt](https://github.com/eried/portapack-mayhem/files/9845188/generate_world_map.bin.txt)

At least this worked well for me, hope it will work for you as well.

eried, 2022-10-22:
Great, you can modify the file directly in github and it will be submitted as a PR. In Windows I had no problems tho, which is strange if it is the line ending

BehleZebub, 2022-10-22:
I'm new to github so thanks for the PR hint, it should reach you now.
Its not about the line endings but integer endiness.

eried, 2022-10-23:
thanks! looks fine :) it will be merged after testing that it works ok 👍

closed 2022-10-28 by GullCode