portapack-mayhem/mayhem-firmware #680: WHIPCALC Clear all my ANTENNAS.TXT file

darknebular, 2022-07-08:
Hello

I have a hackrf 2021 PCB and a Portapack H1.
I have installed the 1.5.4 version.

I put the file called ANTENNAS.TXT into WHIPCALC folder.
Then I run the app of lenght calculator and It shows to me all ants correctly.

When I go to receive ADSB, my file called ANTENNAS.TXT is cleared.

I would like that behaviour doean't exits.

darknebular, 2022-07-18:
I put this .txt called ANTENNAS.txt in WHIPCALC folder: 

#antenna label,elements length in mm, separated by a commas
ANT700,95,134,175,206,230,245
ANT500,185,315,450,586,724,862
TELESCOPICA,110,205,298,391,501

Then I run the app length ant and works fine, when I go to Receivers--> ADSB and come back to the app called Length Ant Calculator I only see the info for ANT500 not the others.

When I open the ANTENNAS.txt files I have this:
baseband_bandwidth=1750000
channel_bandwidth=1
lna=32
rx_amp=1
sampling_rate=3072000
tx_amp=0
tx_gain=47
vga=24
rx_frequency=89000000
tx_freq

It´s changing itself.

darknebular, 2022-07-18:
I proof to change to Windows CRLF, but It´s whipe out too:
[image]

darknebular, 2022-07-18:
With this:
[image]

closed 2022-07-19 by eried

darknebular, 2022-07-21:
The whipcalc file is changed when I enter in ADSB.
Why you close this bug?

eried, 2022-07-22:
I mentioned all these issues in https://github.com/eried/portapack-mayhem/issues/684#issuecomment-1188716291 because they are all about the sd card getting corrupted. Every app that uses the sd card will create problems. 

Imagine I have a screen with a faulty pixel in my portapack, and I open issues about "there is a white dot in the middle of WHIPCALC", "there is a white dot in the main menu", etc...

Can you test another app that writes something to the sd card? if it works properly, then we can reopen the issues.