portapack-mayhem/mayhem-firmware #681: Logs for receives are not readable

darknebular, 2022-07-08:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly verison here: https://github.com/eried/portapack-mayhem/releases)

----

**Describe the bug**
A clear and concise description of what the bug is. 
ADSB or AIS are logging in my root of my microSD with .txt files.
When I open these files I can't read anything with sense...

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Receive
2. Tap on ADBS
3. Read the logs

**Expected behavior**
A clear and concise description of what you expected to happen.
I would like read the logs properly.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release: 1.5.4
* Latest Nightly release:
* Previous working release:

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.

eried, 2022-07-08:
Post a log here, I don't understand what do you want

darknebular, 2022-07-09:
> Post a log here, I don't understand what do you want

You will need wait 1 week for this info, now I am not in my house, due my vacations.

The problem is the log generated is not readable, is full of characters ASCI but without sense...

darknebular, 2022-07-18:
Look, The log files has first lines the lines that I put in ANTENNAS.txt, and then characters without sense:
[image]

darknebular, 2022-07-18:
AFSK.log is weird too, first lines having the lines of the ANTENNAS.txt content and then characters without sense:
[image]

darknebular, 2022-07-18:
I leave empty the logs files and check again, now I can see the logs fine:
[image]

The problem looks like that is when you create a new file.

darknebular, 2022-07-18:
Again, I power on the device again and the logs are generating without sense:
[image]

darknebular, 2022-07-21:
This is an other problem.
Why is closed?

jLynx, 2022-07-21:
Opening again as you said it's not an ad card issue

darknebular, 2022-07-22:
Well, sorry for all, the problem was with a fucking SD-MicroSD Adapter.
This adapter malformed the data...
Now It´s working fine.

Best regards.

closed 2022-07-22 by darknebular