portapack-mayhem/mayhem-firmware #682: PortaPack H2 - Unable to Boot when SD Card is Inserted / Sits at Black Screen

FroggMaster, 2022-07-15:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly verison here: https://github.com/eried/portapack-mayhem/releases)

----

**Describe the bug**

- My PortaPack H2+R3 will only boot without an SD Card inserted. 
- If I have an SD Card inserted The PortaPack H2+R3 will sit at a black screen, no matter which boot configuration/button combination I try when booting.  
- It will not boot with either the left or right button directions when the SD card is inserted
- The screen flickers briefly when trying all of these button combinations but never displays the Mayhem GUI 

I am using a 16GB MicroSDHC SD Card, formatted to FAT32. It has been confirmed functional on multiple devices. 
I have also tried other SD cards but the behavior is the same no matter the SD Card I try. 

- I have already reviewed https://github.com/eried/portapack-mayhem/wiki/wont-boot and none of the suggested fixes/button combinations correct the problem. 
- 
**To Reproduce**
Steps to reproduce the behavior:
1. Flash the latest Mayhem Nightly or Mayhem Version 1.5.4
2. Insert SD card into Portapack H2+R3
3. Attempt to boot with the Encoder Knob and Left Button > Sits at Black Screen
4. You can repeat these steps with the Right button and you will get the same behavior
5. Remove the SD Card, the Portapack H2+R3 will boot without any issues. 

**Expected behavior**
When my SD card is inserted, I would expect the Portapack H2+R3 to display the GUI and Read the SD Card. 

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release: 1.5.4
* Latest Nightly release: Revision (n_220609)
* Previous working release: Modified Firmware with the CPLD check disabled. (1.4.3) 

**Additional**
- It would seem there is some information already present within https://github.com/eried/portapack-mayhem/wiki/PortaPack-Versions indicating that this is somewhat of a known issue. 
- Discuss with @jLynx on Discord regarding some of the behavior presented while attempting to boot. They mention that this may be a timing bug in the firmware. They believe that increasing some of the delays may correct the problem. 
I've included a video: https://drive.google.com/file/d/1VsuhpTyizGgNnbHxbLrYglPOz5vKiH6w/view?usp=sharing
If the bug is difficult to explain, additionally to the text please include images and videos.

jLynx, 2022-07-15:
Thanks for creating this issue. I'll try get to this when I find some time. I am pretty sure I know what the issue is, since it works without and SD card and fails with an SD card, I am pretty sure it's a timing issue since with an SD card it takes more time to load the logic. Increasing some of the delays I added previously should fix this. So I'll do that when I get some time

belkone, 2022-12-02:
hi @jLynx I also have the same problem :(

gullradriel, 2023-04-26:
As no one is talking about this problem anymore the fixes included in latest nightly and latest stable are doing the job.

closed 2023-04-26 by gullradriel