portapack-mayhem/mayhem-firmware #683: Audio Quality with Microphone Application (ui_mictx)

Abraxas3d, 2022-07-17:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly verison here: https://github.com/eried/portapack-mayhem/releases)

I built the latest firmware and installed it successfully (15f3d81)

----

**Describe the bug**
I'm adding modes to the ui_mictx application. First comes learning and using the existing code. In attempting to use FM transmit, the audio quality received from HackRF/H2+ Mayhem on an HT (446 MHz) is poor. It's overdriven and splattery. Turning the microphone gain down (no red on the bar graph) didn't help. Using any number of several different microphones didn't help. Trying another HackRF+H1 PortaPack with the same code didn't help. The roger beep sounds great, but it's synthesized, and not coming from the microphone jack. Turning down the bandwidth to 5 and then 3 and then 2 didn't help. On a spectrum analyzer, in SSB mode, the signal is much wider than it should be. In FM mode, one can see splatter. 

What settings am I missing to eliminate the splatter and/or over-deviation?

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on HackRF+Portapack
2. Select Microphone App
3. Tune Frequency to 446.000 MHz
4. Obtain receive radio. 
5. Listen to the audio. 

**Expected behavior**
I'm expecting the audio to not have splatter, or be wider than expected.

**Affected versions**
I have only tested this with the version that was the most recent (15f3d81)

**Additional**
Thank you for such great firmware! I am happy to help fix or improve FM however I can. But, I want to make absolutely sure I'm not using the firmware incorrectly first.

Abraxas3d, 2022-07-26:
Turning off the micboost (1 = 20dB and 0 = 0dB) greatly improved the audio quality.
There's a background hiss present whether the microphone is plugged in or not, on transmit. Noisy audio circuits? 

Turning off the micboost, and setting mic gain to 1.5 and transmit bandwidth to 15 kHz, achieved "baby monitor quality" audio. This is a big improvement over the firmware when micboost was on.

The goal is to implement a high bitrate voice protocol, so high quality voice from basic FM is something we need to be able to show in order to have a baseline. We are not there yet.

Ideas and guidance very welcome.

Brumi-2021, 2022-07-26:
Hi , long time ago , when I had no H2 WM8731 unit to test and we had to low radiated RF  energy in SSB and DSB I tried to solve it , amplifying the modulator side (and it improved +12 dBs  )  , but that produced the need to use the MIC gain var reduction =0.5 in that WM8731 device.
  Then I had already  pending in my TODO list in this summer  to re-adjust all audio path with MIC gain = 1   in the WM8731  platform and after, readjust again AK4951 platform to also achieve a good levels with MIC GAIN =1 and good radiated power and good VU levels ... 
Anyway , there are many types of MICs , and I was affraid that reducing 20dBs in WM8731 would be too much ...  (as you know in AK4951 we can adjust the gain at any value +1,+2,+3,.. ) . But at the end we will need a similar level at the input to the modulator  .

Abraxas3d, 2022-07-26:
Thank you. I'll help however I can. 

I did a very small change. I forked and then changed the .micboost value from 0 to 1, in the place where values are written to the audio codec. This is in firmware/common/wm8731.cpp

I put the resulting bin file at the top level. 

https://github.com/Abraxas3d/portapack-mayhem

Abraxas3d, 2022-09-06:
Update - 

I'm going to put a probe on the microphone input to see what the signal looks like.
In looking for something in the code that might explain the poor audio quality, I walked through how the WM8731 is set up.
The defaults in the Sampling Control section in wm8731.hpp have SR = 0100. This is an invalid code according to table 18 in the datasheet. This register controls the ADC and DAC sample rates. 
There are different values set in wm8731.cpp, but they are commented out. I think the defaults are being written.
I don't understand the entire codebase, so maybe Sampling Control is commented out in wm8731.cpp because it's being handled elsewhere? 

More soon,
-Abraxas3d

Abraxas3d, 2022-09-06:
Progress. Setting all bits of SamplingControl register to 0's improved the voice quality. 

However, the level is too low and there's a background hiss. The hiss was there before and is still there. The audio levels being low might be something that can be fixed by putting the mic boost back in. I'll try that next. 

-Abraxas3d

Abraxas3d, 2022-09-06:
Mic boost = 20dB is way too hot, but setting mic boost = 0dB and microphone gain (in the UI) to 2.0 and SampleControl to all 0's resulted in the best audio quality on transmit so far. 

-Abraxas3d

Brumi-2021, 2022-09-06:
Hi @Abraxas3d , sorry for still not follow it . (but not forgotten) . As soon as I can , I will have a look on it . 
I know that currently que have to use GUI Mic Gain =0.5 in WM8731 in my device (-6dBs).   My first test  will be, to  just try to reduce internally the WM8731 captured samples by a factor  Gain 1/2, losing 1 bit  , to allow a center usage of a  GUI Mic Gain =1 . (I think we captured 16 or 24 bits from Mic, and later Hackrf just processing 8 , as far as I remember , t.b..c , then it should not be no problem  )
I think that disabling WM8731 mic boost pre amplifier -20dBs will be too much decrease, not good option , and S/N will be worse   (just my feeling) cheers

Abraxas3d, 2022-09-07:
Thank you Brumi. We will get this working really well, I have no doubt.

With the mic boost enabled, the audio is badly distorted (overdriven). Microphone levels are well into the red. It doesn't seem to be the microphone. Two different brands acted the same way. 

If we can fix the gains elsewhere, then turning the boost back on would be good. 

Writing all zeros to the SampleControl register made the most difference, so I'll keep working on that. All zeros is the recommended default, but that isn't necessarily the best settings for the hardware. 

We're going to put a probe on the microphone input signal to see what it looks like. And, we'll try and verify what the Rmic resister value is. That resister sets the first stage gain for the microphone input. The second stage is the microphone boost (0 or 20dB).

Something that we noticed is that the HackRF+PortaPack is microphonic, so there is going to be some level of noise no matter what. You can verify this by unplugging the microphone, pressing transmit, tapping the HackRF+PortaPack case firmly, and listening to the FM signal on something like an SSB receiver. Not too bad. 

-Abraxas3d

Brumi-2021, 2022-09-07:
Hi thanks .
Two small questions .
1-) now in current sw , when you put mic gain GUI at 0.5 it is ok modulation level for you ? (My understanding is yes)

Previous version 1.3.x can be a good  FM reference modulation level to us? 
cheers

Abraxas3d, 2022-09-07:
For the current firmware (1.5.x) all of the mic gain UI settings produce distorted FM audio. It sounds overdriven regardless of the mic gain UI setting.

Writing all zeros to SampleControl and setting mic boost to 0dB produced undistorted audio. Setting mic gain UI to x2.0 gave the best results. 

I have not yet tried 1.3.x, but I can as soon as I get a chance.

Abraxas3d, 2022-09-07:
Proposed Sample Control register writes:

| Bit  | Field Name | Value | Notes |
| ------------- | ------------- |------------- |------------- |
| 0 | USB/NORMAL | 0  | Normal mode because 256 fs MCLK (from clock_manage.cpp) |
| 1 | BOSR | 0  | 256 fs MCLK from clock_manage.cpp |
| 5:2 | SR[3:0] | 0000  | 48kHz audio, 12.288 Fout, 256 fs |
| 6 | CLKIDIV2 | 1  | MCLK is divided by 2, according to ui_mictx.cpp |
| 7 | CLKODIV2 | 0  | kind of unsure here but I don't think we want to divide by 2 again for CLOCKOUT |

Setting CLKDIV2 to 1 brought the audio level up some.

Brumi-2021, 2022-09-11:
Hi @Abraxas3d , 
I was starting to check , that point :  audio sound quality in the transmitted  Mic App , with WM8731 audio codec , yesterday.

My first impressions are the following ones ,  you are right . 
Buf so far , we can still using the Mic app, talking from  a certain distance 

 current  latest nightly fw 1,5.x, using  Mic App ,deviation 75 Khz ,   shows

* Only good  demodulated FM correct audio sound quality , if we are talking into Mic app ,  
     from the mic -mounth distnce , > 10 -15 cms , (and better if we still reduce the GUI  Mic Gain = 0.5)
     
* And high distortion , when we talk  closer to the mic < 10 cms .  (and obviously high distortion when Mic Gain >=1.0 ..) 

I also checked a previous fw 1.3.1 ,(that had only FM tx )   and  it is almost  same or slightly better -
Then it is not a sw regression . 

------------

As  I could confirm that the distortion seems to come from the first  ADC stage , (excesive audio gain in the Mic input) 
Checking  WM8731 datasheet ,  probably the best way to solve it , would be , just  increase value Rin  or add small SMD resistance in series  into the board, 
[image]

I could not find H2+ Portapack schematics , but it should be easy  , just checking the PCB , as it is just the input. 
(but if someone can share the Portapack H2+  PCB v3.6 , schematic  would be appreaciated ) 
[image]

Anyway as changing  1 SMD resistance value in the H2/H2+ Portapack (WM8731 area)   would be not so "friendly-user" to all of us ,  I will start to investigate how to improve it by SW,  giving to the user some option inside the menu been able to  select several  different register settings  to  overcome it or minimize the problem . (as I did in the PR of the AK4951) , pending to be  merged  , PR  #664  .)

Then , let's continue investigating !!!

Brumi-2021, 2022-09-13:
Hi @Abraxas3d , 
I have some simple idea,  to allow to the user to select several lower gain GUI options in WM8731 audio codec. respect (current reference 0dBs, that it is easy to saturate  when talking <5cms to the mic) , 
My current plan , 0dBs(current) , -2dBs,-8dBs ,- 14dBs. (So in total 4 options) I will  try on next weekend . 
I think it will work well , and solve your pointed issue. 
Cheers

Brumi-2021, 2022-09-25:
Hi @Abraxas3d ,   
In order to give maximum flexibiity i was trying  to offer in the Mic Menu , several  mic gain adjust  before ADC , 
keeping the original wm-mic-boost on (+20 dB's) easy to saturate ADC +  3 or 4 options more when wm-mic-boost is OFF (+0dB's)  

But  I am strugling  when tryiing to make  a dynamic adjustment of the the WM MIC BOOST ON/OFF .

Then at the moment I created a test version , with 4 x options to adjust the GAIN , but always with WM MIC BOOST OFF .
Pls. feel free to check and  feedback here,  I will also upload it to the discord . 
It is just and adavanced beta version (pending to check that there is no side effect in the ALC feature in the AK ,) 

Cheers, 
[portapack-h1_h2-mayhem_test_wm8731_mic-gain.zip](https://github.com/eried/portapack-mayhem/files/9640681/portapack-h1_h2-mayhem_test_wm8731_mic-gain.zip)

Brumi-2021, 2022-10-01:
Hi @Abraxas3d , (and all) 

I am glad to finally , been able to offer a similar Mic-ADC gain pre-adjustement   solution for  WM871  audio codec
 than the one that we had for  the AK4951 in the MicTx App.  I struggled  a little , but eventually , I think we reached a quite interesting audio mic  feature !!!!!   (Specially for  the ham amatteur operators ! ) 

The idea was simple,  but as always , I need to try multiple  code test, till reaching a good level.

In AK4951 , I already submitted a PR in May , adding  mic  Auto Level volume  Control , but as it was not merged, 
I will remove it , and add that Combined one , 

(1) When detecting WM8731 audio codec --> allowing Mic Pre-amplifer BOOST (+20 dB's) adjustment .
(2) When  detecting  AK4951 audio codec -> allowing  to play with mic  Auto Level volumen in different max. dB's .

For WM , theoritically we should  have only  two binary togle selection  (+20 dB's / 0 dB's)  , but  as the jump  is  too  much , 
I added a "tricky simple idea"   to allow  5 x lower dB's step  options , and i think they worked quite well !!!! 
My advice , is  
a-) first  --> select in the WM platforms , the suitable "Boost" Option to not saturate the ADC ,
b-) and later, select the post scaling audio samples with the "MIC-GAIN"  : 0.5,  1.0, 1.5, 2.0 
  
Please check in both platforms  ,WM8731  Audio codec , and  AK4951 and  feedback here or in Discord --> Mic App section .
Later I wil submit that PR , I already tested in both platforms , and so far , no problems .
But  the merging also depends on your testing . 

I encourage , specially in the WM Users --> to have a try. 
With that FW hopefully we can solve that audio quality  distorted sound ! 

Cheers, 

[portapack-h1_h2-mayhem_WM_mic_boost_and_AK_ALC.zip](https://github.com/eried/portapack-mayhem/files/9690624/portapack-h1_h2-mayhem_WM_mic_boost_and_AK_ALC.zip)

Brumi-2021, 2022-10-01:
[image]

Brumi-2021, 2022-10-04:
Hi @Abraxas3d 
Whenever you can , pls try this advanced beta test binary . I think you will be happy with it .
Waiting for your news . 
Cheers

Brumi-2021, 2022-10-06:
Hi @Abraxas3d ,
Thanks for your answer . Glad to hear from you again ! 
I added 5 boot options to the Mic app , 2 x on  , 3 x off .
With it , I am sure that you can adjust the Mic sensitivity and select the proper one , based on your mic distance and voice loudness to avoid Mic ADC saturation and distortion . 
Waiting for your test results. 
Hopefully we will close that issue merging it . 
Cheers

Brumi-2021, 2022-10-18:
Hi @Abraxas3d ,   my above  PR has been already merged, and fully tested and validated.
I am  sure that this issue (in your WM8731 audio codec platform )  is solved , therefore that issue , can be closed now.
You can try it in latest ninghtly fw version . 

Cheers,

Brumi-2021, 2022-11-22:
Hi again , this one ,  it is 100% solved .
@Abraxas3d opened it at July , and at that time 
 it was so easy to saturate the mic in WM8731 

But we fixed it , offering  to the user 5 mic boost steps , to adapt to multiple mic gain varieties !

I do not know,  if @Abraxas3d have already have any  chance to confirm it ? ????  

In my opinion , it can also be close it !  Many of us tested it .

Brumi-2021, 2022-11-25:
Hi @Abraxas3d ,  let me close that issue . (the PR was merged already) 
Anyway  , whenever you can , we will appreciate to get your feedback and  confirmation here . 
But I am sure that now, you will find in your WM8731 device  a proper   "mic-boost" gain option , adequate to your mic sensitivity to not produce ADC saturation , and transmit a clean audio quality  .

closed 2022-11-25 by Brumi-2021