portapack-mayhem/mayhem-firmware #684: SD card not mounted after press reset button

darknebular, 2022-07-18:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly verison here: https://github.com/eried/portapack-mayhem/releases)

----

**Describe the bug**
SD card not mounted after press reset button

**To Reproduce**
Steps to reproduce the behavior:
1. Power on and see that the SD is ok and mounted
2. Press reset button
3.  Check it out that now the SD memory card is not mounted

**Expected behavior**
I would like that after do the reset the memory card must be loaded and mounted.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release: 1.5.4
* Latest Nightly release:
* Previous working release:

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.

eried, 2022-07-19:
I think this is something on your unit, probably a bad sd card. Try to format it or with another. 

The errors you are reporting https://github.com/eried/portapack-mayhem/issues/680 https://github.com/eried/portapack-mayhem/issues/681#issuecomment-1187030948 https://github.com/eried/portapack-mayhem/issues/685 https://github.com/eried/portapack-mayhem/issues/679 indicate that it is related with this.

I am going to leave this issue open and close the other ones.

darknebular, 2022-07-21:
I tried with other SD card, I tried format in FAT32 again...

The problem with the reset persists.

darknebular, 2022-07-21:
The error that I reported are an other problems.
I don't think are related.
The SD in initial start works fine, but not the problems that I told you.

Why you close my bugs??

jLynx, 2022-07-21:
All the errors you reported are related to the SD card or sd card storage (corrupted files)

jLynx, 2022-07-21:
So can you please list a few things for me.
What is your current SD card brand and storage size?
What is the alternate sd card brand and storage size?
What portapack are you using?
Does this still happen on the latest nightly?

darknebular, 2022-07-22:
I have two portapack H1-R2. They have got speaker conector (now the conector is un-plugged). I tried with a another micro-SD from Kingston 16GB 100MB/s Cs=10.
The problem with the loading MAP, my whipcalc file and my lookingglass file still changed when I open ADSB app.
The problem of unmount after do a reset persists.

darknebular, 2022-07-22:
Well, sorry for all, the problem was with a fucking SD-MicroSD Adapter.
Now It´s working fine

closed 2022-07-22 by darknebular