portapack-mayhem/mayhem-firmware #689: Peut on rajouter des fonctions ? issus du firmware MAHO  ?

doc62fr, 2022-07-23:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen. Remember that adding stuff is always possible, but time is a limited resource for everyone. Check the wiki for more information how to compile the firmware and try to explore modifying the code yourself.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional**
If the suggestion is difficult to explain, additionally to the text please include images and videos.

doc62fr, 2022-07-23:
il serait intéressant de rajouter quelques applications issu du firmware Maho

voir ici : https://github.com/jamesshao8/portapack-havoc-modified

certaines existe déjà , mais d'autres sont oublier...

merci de votre retour

doc62fr, 2022-07-23:
on pourrais par la suite ajouter :

Affichage VOR et ILS
Affichage coordonnées GPS (deja prevu dans le firmware SAHO
Bluetooth

comme sur les Icom IC-A25... ou les Yaseu FTA-850L  (anciennement FTA-750L et FTA-550L)

Aux travail les Developpeurs et Programmeurs  :+1:

gullradriel, 2023-03-27:
Le dépôt git de ce monsieur est un dépôt qui ne contient pas ses sources publiques.
Il ne respecte pas la GPL.
Les développeurs ne peuvent pas faire grand chose, mais vous pouvez contribuer.
Nous attendons votre Pull Request avec impatience.

closed 2023-03-27 by gullradriel