portapack-mayhem/mayhem-firmware #693: Meshtastic and LoRa functionality

IIMacGyverII, 2022-07-27:
Would it be possible to make a feature allowing Meshtastic or other LoRa communication?  Meshtastic is open source and it may be pretty easy to integrate in to the awesome portapack-mayhem firmware.  Any interest?

Here is a link to the Meshtastic repo.
[https://github.com/meshtastic/Meshtastic-device](https://github.com/meshtastic/Meshtastic-device)

Sliicy, 2024-05-14:
This in theory would also be able to hop between different frequencies used in different countries, like 915 in USA, 433 or 868 in Europe, etc...
I also would love this functionality.

nbirdmann, 2024-11-04:
Maby this would work with the usb esp32 addon made by htotoo

gullradriel, 2024-12-04:
Duplicated issue. Tracking is in https://github.com/portapack-mayhem/mayhem-firmware/issues/400.
Closing.

closed 2024-12-04 by gullradriel