portapack-mayhem/mayhem-firmware #694: limitation aux  99 premieres fréquences dans le scanner

doc62fr, 2022-07-28:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen. Remember that adding stuff is always possible, but time is a limited resource for everyone. Check the wiki for more information how to compile the firmware and try to explore modifying the code yourself.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional**
If the suggestion is difficult to explain, additionally to the text please include images and videos.

doc62fr, 2022-07-28:
bonjour

peut on prévoir pour aller au dela des 99?

merci de votre retour

73

eried, 2022-07-29:
hei, du må skrive issue-teksten på engelsk

closed 2022-07-29 by eried

doc62fr, 2022-08-07:
limitation to the first 99 frequencies in the scanner radio!

can we plan to go beyond the 99?

my frequency files have more than 99 frequencies, I only have access to the first 99 during a scan