portapack-mayhem/mayhem-firmware #700: Bias DC Voltage not working in TPMS & ERT

bluegizmo83, 2022-08-07:
Running the latest v1.5.4 firmware. Bias DC voltage is working in all Receive apps except TPMS and ERT Meter. It doesn't output voltage when DC voltage is already enabled, and toggling the DC voltage doesn't do anything when using TPMS or ERT Meter.

Brumi-2021, 2022-08-08:
Hello , yes we knew about it . But so far I did not see any LNA for TPMS or ERT . Which set up are you condidering for those two app’s ? 
cheers ,

bluegizmo83, 2022-08-08:
> Hello , yes we knew about it . But so far I did not see any LNA for TPMS or ERT . Which set up are you condidering for those two app’s ? cheers ,

The LNA I'm using is a wideband LNA, the Nooelec LaNA, it works from 20MHz to 4GHz: https://www.nooelec.com/store/lana.html

Brumi-2021, 2022-08-09:
Hello , ok , checking the current specs of the link , it tells 50 to 80 mA , is in the limit of the hackrf (<=50 mA as far. As I remember) . But it should be ok if you already tested it in another mayhem app’s.  
 TPMS I think it is in the 315 or 433 MHz depending on region . But I do not know about ERT frequency ? (Just to learn)
 
It should not be  difficult to activate also the DC bias control in those two applications. In fact I added in several other apps , I just did not do in those two , because I did not see the need or demand . 

If no one solve it , I can do it in next week/s. 
Cheers

bluegizmo83, 2022-08-09:
> Hello , ok , checking the current specs of the link , it tells 50 to 80 mA , is in the limit of the hackrf (<=50 mA as far. As I remember) . But it should be ok if you already tested it in another mayhem app’s. TPMS I think it is in the 315 or 433 MHz depending on region . But I do not know about ERT frequency ? (Just to learn)
> 
> It should not be difficult to activate also the DC bias control in those two applications. In fact I added in several other apps , I just did not do in those two , because I did not see the need or demand .
> 
> If no one solve it , I can do it for you in next week/s. Cheers

Ok, thanks!
Also, I think ERT frequency is in the 915MHz band if I recall correctly.

Brumi-2021, 2022-08-09:
Ok , no problem , understood . 
I will try to fix it in both remaining apps in next days .
Cheers

Brumi-2021, 2022-08-21:
Hi @bluegizmo83 ,
Pls. find attached the advanced binary test version , adding the requeted  ANT-DC-BIAS also to the ERT (911,6Mhz)  and TPMS (315 / 433 Mhz) Apps .

I will submit  in parallel  the related PR,  but we will appreciate your test evaluation feedback , 

Note : recently from the PR #662 onwards,   sometimes I have problems for testing and debugging using my local compiled binary .  Sometimes , it keeps on  booting with  black screen , and I can solve it with DFU and later reflashing with any old version 1.4.x and  then come back to the latest one  "next" . But when it works well , in my H1 device -using my local binary -  the App Settings can not be activated in my device (it may happend to you, using that binary, not sure (?)   .    (but when using official  web binary release,  always woks well the boot and the App settings . Then I believed that when merged all will be ok) .  

Cheers, 
[portapack-h1_h2-mayhem_700.zip](https://github.com/eried/portapack-mayhem/files/9389654/portapack-h1_h2-mayhem_700.zip)

bluegizmo83, 2022-08-22:
@Brumi-2021 That compiled firmware you linked above worked great for enabling ANT-DC-BIAS for ERT and TPMS, thanks! I also had the black screen of death after flashing this firmware and had to DFU recovery to get it working again. However, I also had to do the same thing when I first flash the officially released Mayhem compiled firmware as well, so its not just you or your compiled firmware! I thought it was just me and that I did something wrong when I first flashed my Portapack H2, but since you also had this issue, there must be something wrong in the official firmware causing this...

jLynx, 2022-08-22:
@bluegizmo83 you need to read the wiki regarding the black screen. It's perfectly normal and dfu not required

bluegizmo83, 2022-08-22:
> @bluegizmo83 you need to read the wiki regarding the black screen. It's perfectly normal and dfu not required

I believe I tried the "holding LEFT or RIGHT or UP or DOWN or OK/SELECT on the DPAD while booting" and that didn't help back when I had trouble the first time. I could be mistaken though as I should've remembered to try that this time if I had done it before, but I didn't... I will remember to try that next time though! Thanks

Brumi-2021, 2022-08-23:
Hi @bluegizmo83 , thanks your validation (it is a binary derivated from current "next").

Hi @jLynx , I already started to mention you about that strange issue to you on 15th of june . (Private chats)
But now I have some additional information . I found that any version that I compile locally  including PR #662 , sometimes produces a black screen in my H1(standard original QFP64 Portapack CPLD) , brick (no way to recover following your won't boot instructions ) and the only way that I found to unbrick it is DFU and reflash it with any old version prior #662 (ex  version 1.4.x) and then reflash it again with my above binary . But in that case , it powers up well -correct screen-, but I can not store anything in the app settings . And when checking the .txt file inside SDcard /hardware it says "0". But i am almost sure that if you compile this PR , your created binary would be also ok ....(same as your 1.5 4)

Pls try my above binary in your device and let us know about it . Maybe it is due to my compile chain set up (gcc-arm-none-eabi version 10 3)? ...or some timing too critical in the #662  but there are something critical in execution time ,  in my created binaries that may produce those strange things . I need to continue debugging it ... 

By the way , I also tried to play increasing the 8 delays that you added in the Portapack.c till 300msegs and no difference...  any further hint or clue would be appreciated. 
Cheese

Brumi-2021, 2022-09-05:
Hi , then , now that PR is merged ,
this issue can be closed ! 
Thanks

closed 2022-09-05 by bluegizmo83