portapack-mayhem/mayhem-firmware #708: Morse TX: no modulation in CW mode

PolymerPrints, 2022-09-02:
**Describe the bug**
Trying to transmit a Morse message in the Morse TX application with "CW" modulation. The transmitted signal is simply a single tone/carrier, there is no modulation. In FM mode everything works as expected. Signal checked using an RSP1A and SDRUno.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Morse TX.
2. Set modulation to CW.
3. Set any message.
4. Press "start".
5. Monitor transmitted signal.
 
**Expected behavior**
Expected to see a modulated Morse signal, not a single tone.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release
* Latest Nightly release

Older versions not checked.

TekNicholas, 2022-09-04:
I just tested this and cannot reproduce this bug.  Used an ICOM IC-705 on the 2m band in CW mode for reception and it picked up the signal.  There was a quieter echo of the signal around 300ms after every dit and dah that made discerning the characters a bit more difficult, but the main signal was clearly there.  Tested at different WPM speeds, tx powers and different distances from the receiver with no change.

PolymerPrints, 2022-09-05:
Thanks for trying to reproduce the issue! I spent some more time trying to get it to work on my end, but unfortunately to no avail. At this point I'm out of ideas of what to try next; if more people can confirm the issue is not reproducible I will assume the issue is on my end and close the issue.

Brumi-2021, 2022-09-05:
Hi @PolymerPrints , do you have any other modulation problem with the rest of TX App ? (Example Mic FM, AM  has modulation? ....) if you have strange TX modulation issues,  I may  address to another possible  solution.  
It is rare , but we had it 
Cheers

PolymerPrints, 2022-09-05:
Confirmed working TX applications: soundboard, SSTV, morse with FM modulation, replay of Tesla sample file (which I believe is AM). If you have a suggestion of another test I can perform, please let me know!

Brumi-2021, 2022-09-05:
Hi @PolymerPrints , 
As you can see below,
 I remember two Problem issues linked to some abnormal Tx modulation behaviour . Both were solved reflashing again Hackrf CPLD flash . It may happen that some clone vendor  produced some batch wrting a corrupted Hackrf CPLD flash .(that is my guess)

Try to follow my 22nd of March commented here on  #466 .
(Later on , other colleague also reported another Problem issue #553 about Morse was also fixed with the same solution ) 

good luck , and pls confirm it .
If you need extra help, let us know about it . 

Cheers

PolymerPrints, 2022-09-07:
@Brumi-2021 Thanks you for the response!

Indeed I found your post a couple days ago and I flashed the HackRF CPLD (taken from the latest firmware from the HackRF git). Unfortunately it did not fix my issue. Note that I did not disconnect the HackRF from the PortaPack, however the LEDs behaved as expected (update successful).

I also tried to downgrade to a 1.4.x release to see if it would work, but I could not get my PortaPack H2 to work properly (screen stays black, pressing directional buttons did not affect it).

Brumi-2021, 2022-09-07:
Hi @PolymerPrints , thanks for testing it, just in case .
So far I still never played with Morse app , but let me try later, to see if I can also reproduce that problem or it is unique to your device  . 
Cheers

PolymerPrints, 2022-09-07:
@Brumi-2021 Thanks! I have tried it at many different frequencies, but mostly around 432MHz.

Brumi-2021, 2022-09-07:
Hi @PolymerPrints 
I checked with old version fw 1.3.0 under H2+ WM8731 ,   and I think it works correctly .
When selecting Morse App , Modulation :  FM   ,  then the selected tone freq (ex. 700 Hz)  is used to modulate the text in morse .
Morse code symbol is formed by a sequence of dits and dahs. (The duration of a dah is three times the duration of a dit. )
But in this case , each symbol is  marked by a modulated tone .
[image]

But when you selecting CW ,  the tone modulation is ignored (just sending carrier ) and you just got the same transmitted  message showing   large  and short symbols but this time each symbol is marked just by just carrier ON duration (same as you will do by a light on/off to send morse msg . You are using just CW (continuous wave and the receiver will add a local oscilator mix to  get an audio tone .
[image]

Then to my eyes , is working weill , just  maybe confused GUI ?
Which fw version you pointed , last nightly ?  How about 1.3.0 example ? 

Cheers,,

Brumi-2021, 2022-09-07:
Hi @PolymerPrints , 
in your first attached picture you have a continuous Carrier signal always ON ? Or it is making ON/Off carrier  ...? 
if always ON, so, then it is a real problem , and i will check (in next days) my same unit with the latest FW next . 

By the way, regarding your comment „about getting black screen , it does not matter the pressed button , I am suffering exactly the same problem . (I already comment to @jLynx in my last chat inside #700 
(Pls check it , I am starting to consider that I am not the only one )

EcstaticMark, 2022-09-07:
Hi, all.
I'm having the same problem with my Portapack and Hackrf One, exactly as described in the first message. I was able to transmit  (and decode) CW with the Hack RF using SDR Angel. I can see the carrier go on and off as expected. But from the Portapack all I get in the CW mode of Morse is a carrier that stays on for the duration of transmission.

PolymerPrints, 2022-09-08:
@Brumi-2021 Sorry, should have been more clear! The first image shows to separate transmissions, one completed and the other in progress, so two always on carriers.

Brumi-2021, 2022-09-09:
Hi @PolymerPrints and @EcstaticMark ,
Which are your exact Hardawre unit and fw where you find that  problem  ? 
Maybe not related , but just in case ,  I want to know your unit conditions : 
H1/H2 ?
, AK4951/WM8731?
 big CPLD QFP100 or small QFP64 Portapack? 
And fw version that shows that problem ?

EcstaticMark, 2022-09-09:
Hi, Brumi-2021.

H2+, Mayhem V 1.4.2 (from menu).

hackrf_info version: git-7047252
libhackrf version: git-7047252 (0.6)
Found HackRF
Index: 0
Serial number: 0000000000000000f77c60dc28822ec3
Board ID Number: 2 (HackRF One)
Firmware Version: local-6c4dec59 (API:1.04)
Part ID Number: 0xa000cb3c 0x00584f67

If you tell me or send me a link to a page that explains how to tell AK4951/WM8731 and big CPLD QFP100 or small QFP64 Portapack I'll get those or any other info for you..

Brumi-2021, 2022-09-09:
Hi @EcstaticMark , thanks 
To check àudio codec IC , it is easy just from the menu, i think going to debug menu , and peripheral-not sure - and you will see an icon about the audio codec IC that you have in your Portapack board .clicking there you will see the register settings values. 
To check CPLD type of the Portapack, you need to see the Portapack brd it self visually. But do not worry , maybe not relevant for this problem . I have both types and I will check in both .

EcstaticMark, 2022-09-09:
Bruni-2021, it's a WM8731. I have the unit disassembled so I'm game for identifying the components.

Brumi-2021, 2022-09-09:
Ok, to check CPLD , is that Ic , just to know if you have TQFP100 pins (big one , second photo) or TQFP64  (original CPLD, as blue colour board)
[image]
[image]

EcstaticMark, 2022-09-09:
> [Screenshot_20220909-135041_Docs](https://user-images.githubusercontent.com/86470699/189344410-e879760e-4742-4d3f-99e6-51537a9d808b.jpg)
> [Screenshot_20220909-135229_Gallery](https://user-images.githubusercontent.com/86470699/189344423-f5cfa691-51c1-4ce7-ae44-03143cb3c4dc.jpg)

TQFP100, right down to the round holographic sticker.

Brumi-2021, 2022-09-10:
Hi @EcstaticMark , that de 1.4.2 that you comment,
is the one that' you can see in the bottom left from the menu , no ? I mean it is the original shipping conditions of the Chinese vendor, no ? (Probably 1.4.2 jumbo version)

I got a unit exactly same as yours , but I changed many times the fw . In the afternoon I will try that 1.4 2

EcstaticMark, 2022-09-10:
> Hi ***@***.***(https://github.com/EcstaticMark) , that de 1.4.2 that you comment,

Hello Brumi-2021. It is the original firmware, displayed in the lower left.

Brumi-2021, 2022-09-10:
Hi @EcstaticMark 
I tried in your exact same hw (H2+ ,WM8731 , Big CPLD Portapack ) the original shipped Mayhem FW conditions : “fw 1.4.2 jumbo version “ and now in my set with that one, now , I can not reproduce your problem . 

But it is true that I have made , long time ago , another prior software change to my unit to solve other problem . Then we are not checking exactly same fw and Hackrf_CPLD binary conditions , and that could be a good  clue to maybe solve your problem .!

My Chinese vendor told me , that in his line , when he was assembling both boards : hackrf and Portapack, he told me ,that the HackRF boards that they were using , it had pre–flashed a very old Hackrf_CPLD version , not as optimised and correct as the last one (and that could be maybe the reason of your issue?). 

Then i recommend you 
(1) to first check , visually if you also have now  other jerky motion problem (#651 ) . (If Yes , it is confirmed that your hackRF board was originally having a very old fw and old Xilinx HackRF CPLD version) 

(2) and if you also have it (like me in the past  ) then pls try to follow my instructions,explained there , (#651 ) updating first HackRF fw , stand alone fw , and after it , rewrite the same mayhem  fw 1.4.2 jumbo version .

(3) after that test . Pls confirm  If you fixed the Morse CW problem or not ,? And let us know .
Eventually I guess that you may fix both things , jerky motion LCD and that strange CW morse issue ???

(4) and from that stable point , from 1.4.2 jumbo version  , you will be able to reflash with latest Mayhem version1.5,X , taking care about the correct booting driver selection (just the first time) , to avoid black screen. 

But take special care , doing the process , and if you have more questions , pls ask us , and we will try to help you .   Good luck !  Best case , we really found out your strange problem . 

Cheers

Brumi-2021, 2022-09-13:
Hi @EcstaticMark ,
 have you had any chance to disassembly the boards (hackrf one and Portapack)and reflash Hackrf one , and later reassemble and reflash again with Mayhem ? (As I was describing in previous post )

EcstaticMark, 2022-09-13:
Hello, Brumi-2021. It will be a while until I'm ready to update my Portapack, if indeed I ever do.
I'm not the one who reported the issue, I was only trying to help troubleshoot it.

Brumi-2021, 2022-09-13:
Ok no problem , 
yes i know that this problem was open by @PolymerPrints , not you .  But as you could  replicate it  and I have a Chinese unit  with exact  same Hw  as yours , but now I can not replicate it. I am assuming that it is  because I updated long time ago the Hackrf fw part (that automatically updated the Xilinx haxkrf-cpld) , I was just assuming that this fw CPLD version   difference should be the real root cause to solve it .
Anyway let's wait till you or @PolymerPrints can try my proposal.  And see his feedback. 
Sorry to push you , as you could also replicate it ,  I was just trying to solve that strange issue with your help. 
But before update it , can you also recognize in your set the slow jerky motion in the waterfall FFT part? (That is easy to recognize and test )
Cheers

PolymerPrints, 2022-09-13:
@Brumi-2021,

I will try to do the updates soon and report back. Do note however that I do not have a HackRF clone (original one that I updated to the latest version when I got it, not connected to PortaPack) and no issues with the waterfall motion.

Brumi-2021, 2022-09-13:
Hi @PolymerPrints ,  thanks for your answer,
If you do not have a clone Hackrf , and no issue with waterfall motion,  maybe no meaning to test . The root cause should be another one...

But  as I can not replicate it  , no clear idea ...still we are lost ...

What are the common points between your unit and @EcstaticMark ? 

You also have Portapack H2+, WM8731 Audio codec, with big CPLD TQFP100 , with any fw Mayhem ? 
Cheers

PolymerPrints, 2022-09-14:
@Brumi-2021 Our HW is the same as far as I can see, WM8731 & TQFP100, however not sure what the difference is between H2 and H2+. For FW I tried the latest nightly, 1.5.4 and also a 1.4.1 release that was supplied by the PortaPack seller. I'm seeing the issue on all of these firmwares.

Brumi-2021, 2022-09-14:
@PolymerPrints , in my PCB Portapack it mentioned H2+ PCB Version 3.6 and it came with battery and speaker . 
Anyway the point is that I tried to reinstall the previous original seller special 1.42 from jumbo , and it does not fail . ( but it is true , that long time ago I updated the Hackrf fw and I assume that also  his Xilinx CPLD, long time ago when  I found strange motion in the waterdall)

The  that is the point , my same HW than you is not showing now , any problem about CW Morse  with original  fw 1.4 2 from Jumbo

EcstaticMark, 2022-10-11:
Hello again, Brumi-2021. Thank you for bird dogging this down.
Yes, I seem to have the 'jerky' problem.
Problem is, the screen for my Linux box recently died and I'm not sure about doing the firmware updates on a Windows PC. Also, I wasn't even aware of either problem, so I'm not desperate for a fix. Maybe someone else with the same issues can confirm the fix. Once I get a new monitor for my Linux machine I'll revisit this and attempt to apply the fix.

Thanks again.

gullradriel, 2023-04-26:
The issue have been discussed and answers have been given. You may open a new one if it's still here.

closed 2023-04-26 by gullradriel