portapack-mayhem/mayhem-firmware #709: After firmware update to 1.5.4 cannot get to Mayhem

hgroth, 2022-09-10:
I successfully upgraded the firmware to 1.5.4 but the Hack RF does not switch back to 
the Mayhem screen and the display remains dark. I tried to reboot several times and still
nothing happens

SublimeMudTime, 2022-09-10:
Check these steps out:
https://github.com/eried/portapack-mayhem/wiki/Won't-boot

I'm having the same issue. I am about to re-hash these steps but also am going to try downgrading to see if I can recover it.

Brumi-2021, 2022-09-10:
Hi , In case it help you ,
i do not know if we are talking about same issue ??
————————
in some early chats , I also comment some strange behaviour related to the latest official fw version : 

„I found that any recent version that I compile locally including PR https://github.com/eried/portapack-mayhem/pull/662 , sometimes produces a black screen in my H1(standard original QFP64 Portapack CPLD) , brick (no way to recover following your won't boot instructions ) and the only way that I found to unbrick it is DFU and reflash it with any old version prior https://github.com/eried/portapack-mayhem/pull/662 (ex version 1.4.x) and then reflash it again with my above binary version . But in that case , it boots well -correct screen-, but I can not store anything in the app settings“ 

pls share if you could also recover 1.5.4 with my methode , but in that case , can you store user app settings?

mikachuman, 2022-11-28:
H1 portapack. Went through all the no-boot troubleshooting steps and I'm still having issues. I can get to the hackrf mode and can update the firmware for portapack and hackrf but the device either boots on white or black screen - depending which remedy I applied previously (re: LCD drivers). But there's no booting into portapack mode anymore. Tested first with latest nightly and then with 1.5.4. Should I try flashing with 1.4.x as a solution here?

Brumi-2021, 2022-11-28:
Hi @mikachuman if the device boot is in white or black screen ,first from 1.5.4 , I suggest you , -if you did not - to try to follow the boot init , for the first time  https://github.com/eried/portapack-mayhem/wiki/Won't-boot. Did you already tried it before ? (normally when booting keeping pressing the proper push button for few seconds .) 

If you have H1 Portapack , I guess that your Portapack is using standard QFP64 CPLD, not ? Then It should work with any normal previous release (no need special jumbo version ) . (As a additional test , you can also try 1.4.x to see if you got correct boot). If it works , we should fine a way to making work with 1.5.4 also

mikachuman, 2022-11-28:
Hi @Brumi-2021 - thanks for the response. I tested those different LCD driver initialisations with the buttons, but they only produced either white or black screens (immediately and on subsequent boots). This time I tested with 1.4.4 and it works. Still haven't tried if it fails on 1.5.x. Do you think I could update from 1.4.4 to 1.5.x now that I've been on a previous firmware? Or can anyone just push the latest stable on a portapack and it will work on a previously empty system?

In any case for people experiencing the same: my issue was resolved by installing 1.4.4 - might test later if 1.5.x works.

mikachuman, 2022-11-28:
> Hi @mikachuman , I am glad to see that you recover your device .
> 
> from the PR #662 onwards , (included in 1.5.4) sometimes I am having exactly the same problem than you , not booting in the binary that I am compiling (but working well if I am downloading the binary from official web ). (It is my compiler diff version issue) The only way that I found to recover my unit is to reflash it with a previous fw (without PR#662, example 1.4.x), and from that point , I can reflash 1.5.4 without problems . (So to restore the brick I need to preflash 1.4.x before , only once .
> 
> PR #662 , is treating specially the persistent memory of the CPU , that is used in the boot process. But even we have a correct cell coin battery , in my case , sometimes goes to some strange corrupted status, that does not allow booting same as you ... (I did not study yet ) . In your case , it was also using your compiling binary ?

Hi @Brumi-2021,

I used the release artifacts (1.5.4 .zip and 1.4.4 7zip) and installed with the .bat files - didn't compile anything myself. Tested what happens if I try to update firmware from 1.4.4 to 1.5.4 -> same result, black or white screen depending on previous button pressed during init.

gullradriel, 2022-11-28:
Have you tried the nightly ? I can remember some PR's about testing and resetting persistent memory inside.
You can keep the same datas.

Brumi-2021, 2022-11-28:
Do you have cell coin battery in the Portapack board ? 
try also same boot from 1.5.4 , removing the SD card and keep pressing the proper push button in the boot time , any difference ?

mikachuman, 2022-11-28:
> 

Tested the latest nightly first. Was a nope.

Regarding the cell coin battery - I'm not sure if it was provided with one. I don't remember installing one if there was a slot. Would need to find my allen key to confirm that. :)

Earlier I also tested 1.5.4 while removing the SD card with different push buttons during boot. Same result.

mikachuman, 2022-11-29:
Tested further and this issue begins at release 1.5.0.

Persistent memory seems to work since I can boot with down or left and get a black or white screen that persists to the next time I boot. I checked the code and there's so little stuff that maybe the artifacts are somehow broken? A compiler issue?

Brumi-2021, 2022-11-29:
Hi , if you start to have problems from  fw version. 1.50 , then we may need help from @jLynx .
From that version , I think he introduced  to record the detected hw platform in the SD or persistent memory and boot correctly in the next boot . That was introduced to solve the black screen in the big QFP 100 CPLD Portapack version . 
make sure that you have a good voltage in the coin cell battery of the Portapack and correct SD (or test it removing the SD ) . By the way I think that while reset if you keep pressing central button it will try to auto detect the hw platform . 
in fw 1.50 , some users had still boot problems and he was adjusting some boot msegs delay in next versions , fw 1.5.1,   1.5.2,... improve that reliability. 
try with a new cell battery or removing the SD ....
No more ideas ...

mikachuman, 2022-11-29:
> Hi , if you start to have boot problems from fw version. 1.50 , then we may need help from @jLynx . From that version , I think he introduced to record the detected hw platform in the SD or persistent memory and boot correctly in the next boot . That was introduced to solve the black screen in the big QFP 100 CPLD Portapack version . make sure that you have a good voltage in the coin cell battery of the Portapack and correct formatted SD (or test it removing the SD ) . By the way I think that while reset if you keep pressing central push button it will try to auto detect the hw platform . in fw 1.50 , some users had still boot problems and he was adjusting some boot msegs delay in next versions , fw 1.5.1, 1.5.2,... improve that reliability. try with a new cell battery or removing the SD .... Sorry no more ideas by now ...

I'm positive that the cell battery is still working because it stored the previous screen setting (black or white screen) even after the boot. But maybe resetting the memory by taking it out and reinserting it might have an effect. I'll look into it if only to eliminate any odd issues with the battery. The battery is a fresh one from a good brand and the screen driver setting was stored and retrieved successfully after boot, so I'm very doubtful of that being the cause.

Thanks for all the help here @jLynx

gullradriel, 2023-05-10:
No more support on that release.
Please update to latest stable or nightly, and open a new issue if the problem persist.

closed 2023-05-10 by gullradriel