portapack-mayhem/mayhem-firmware #712: scaricare file git hub per scheda sd

Fil974, 2022-09-16:
Salve volevo chiedere come si fa a scaricare i files git hub  per copiarli poi nella scheda sd?
Ma poi li devo copiare tutti nella scheda sd?
O basta la prima cartella git ?
Grazie!!!

eried, 2022-09-16:
No estoy seguro que haya gente que hable italiano acá, creo que es recomendable escribir tu problema en inglés (si te fijas, todo está en inglés y sería extraño que escribieramos en otro lenguaje, no? 😁)

GullCode, 2022-10-09:
Translation:
"Hi I wanted to ask how do you download the git hub files to then copy them to the sd card?
But then do I have to copy them all to the sd card?
Or is the first git folder enough?
Thank you!!!"
-You do not need the git folder except if you're going to make some code/new app/firmware modification and build your own. The portapack does not need it.
-You have a special PACK for the firmware to run smoothly. Download it from here : https://github.com/eried/portapack-mayhem/releases/ , unfold 'assets' and choose the zip file mayhem_nightly_n_DDMMYYYY_COPY_TO_SDCARD.zip
-On that page you can also find the latest firmware files to flash

closed 2022-10-09 by GullCode