portapack-mayhem/mayhem-firmware #715: How AIS format works?

HaXlezi, 2022-09-29:
Hi guys, can someone explain me how does data format for AIS recieving work please? I mean there is txt file with two columns but idk for what purposes are they. Can you help me please

[image]

eried, 2022-09-29:
It is just a dump https://github.com/eried/portapack-mayhem/blob/f161e85f960cff0c166173f4f7a4244b8c625375/firmware/application/apps/ais_app.cpp#L143 I think there is no decoder written for it

HaXlezi, 2022-10-12:
> > Hi guys, can someone explain me how does data format for AIS recieving work please? I mean there is txt file with two columns but idk for what purposes are they. Can you help me please
> > [image]
> 
> Hi how do you extract it ? Is it through GNURADIO on PC screen , or it is the portapack captured LCD screen ? Thanks

It's just screenshot of txt file which should be automatically created on SD card after some AIS receiving

marcelrv, 2022-10-15:
If your question is how to make sense of the string you see... google for `ais online decoder` e.g. you will find https://www.aggsoft.com/ais-decoder.htm

you can paste the string in the 2nd column in there and you see what the message actually means.

eried, 2022-10-15:
> If your question is how to make sense of the string you see... google for `ais online decoder` e.g. you will find https://www.aggsoft.com/ais-decoder.htm
> 
> you can paste the string in the 2nd column in there and you see what the message actually means.

The problem is that the portapack is not saving the sentences but a raw, and I dont think there is written a converter for this

gullradriel, 2023-05-10:
Closed as answered to the best of our knowledge.

closed 2023-05-10 by gullradriel