portapack-mayhem/mayhem-firmware #718: error: invalid application of 'sizeof' to incomplete type 'jtag::tap::{anonymous}::table_t'

StanleyLex, 2022-10-03:
Hello! I ran into the following problem when assembling the firmware.

`/home/user/git-repositories/portapack-mayhem/firmware/common/jtag_tap.cpp:84:15: error: invalid application of 'sizeof' to incomplete type 'jtag::tap::{anonymous}::table_t' {aka 'std::array<jtag::tap::{anonymous}::entry_t, 16>'}
   84 | static_assert(sizeof(table_t) == (16 * 4), "Unexpected size of table_t");
      |               ^~~~~~~~~~~~~~~

/home/user/git-repositories/portapack-mayhem/firmware/common/jtag_tap.cpp:86:15: error: variable 'const jtag::tap::{anonymous}::table_t jtag::tap::{anonymous}::table' has initializer but incomplete type
   86 | const table_t table { {
      |               ^~~~~

/home/user/git-repositories/portapack-mayhem/firmware/common/jtag_tap.cpp:105:35: error: variable 'const std::array<const char*, 16> jtag::tap::{anonymous}::state_name' has initializer but incomplete type
  105 | const std::array<const char*, 16> state_name { {
      |                                   ^~~~~~~~~~

/home/user/git-repositories/portapack-mayhem/firmware/common/jtag_tap.cpp:124:35: error: variable 'const std::array<const char*, 16> jtag::tap::{anonymous}::state_long_name' has initializer but incomplete type
  124 | const std::array<const char*, 16> state_long_name { {
      |                                   ^~~~~~~~~~~~~~~

/home/user/git-repositories/portapack-mayhem/firmware/common/jtag_tap.cpp:124:35: warning: 'jtag::tap::{anonymous}::state_long_name' defined but not used [-Wunused-variable]
make[2]: *** [firmware/application/CMakeFiles/application.elf.dir/build.make:1379: firmware/application/CMakeFiles/application.elf.dir/__/common/jtag_tap.cpp.obj] error 1

make[1]: *** [CMakeFiles/Makefile2:693: firmware/application/CMakeFiles/application.elf.dir/all] error 2
make: *** [Makefile:91: all] error 2
`

Unfortunately, my qualifications are not enough to cope with the error on my own. I ask for help.

GullCode, 2022-10-03:
Hello, what is the version and what compiler have you used ?

StanleyLex, 2022-10-03:
I am using gcc 12.2.0

GullCode, 2022-10-03:
AFAIK you have to use gcc version 9.3.1 20200408 (release) (GNU Arm Embedded Toolchain 9-2020-q2-update) 
When I tried to use the latest toolchain it was not working too.
If you're on linux, then you can scrap the needed parts in the following tutorial from the FAQ:
https://github.com/eried/portapack-mayhem/wiki/Compile-firmware#using-arm-on-debian

Brumi-2021, 2022-10-03:
Hi @StanleyLex ,
in my case, linux pentoo (gentoo) 
We also had many  compile errors with gcc 11.2 .
And after changing back to 10.3_p202110 , all is stable and ok . 

Cheers

GullCode, 2022-10-09:
Closing as it's a compiler version related thing.
You may try to update the code and make a PR.

closed 2022-10-09 by GullCode