portapack-mayhem/mayhem-firmware #720: Reset and Capture hanging

sdb333777, 2022-10-08:
anyone having an issue where at first power on, if i goto Capture it hangs. If i goto one of these 3 - search , microphone, scanner at first power on and then goto capture it will goto it. i have to power off in order to select search first before i goto capture.
i tested it with the 2021 original firmware and mayhem 1.5.4 and still same thing. and still does it with the newest current firmwares
also reset always hangs anytime i press it with the portapack hooked up, but not with the hackrf alone.

eried, 2022-10-08:
hangs = just freezes?

GullCode, 2022-10-08:
I have that too on my H2.
Freeze and LEDs flashing

sdb333777, 2022-10-08:
yes it freezes, watching the clock second counter and pressing reset freezes, only power off and back on gives control back to the bottons or wheel. power on and going to capture first thing makes it freeze as well. someone else i know has this same issue #720 with his unit reset not working and the clock freezing and capture not working at first power on so is it the portapacks or the mayhem firmware?

Brumi-2021, 2022-10-08:
Hi , as we chat through discord .

I have similar device HW than you (special devil)  , and initially I had many incidences and hang ups , (and this issue #651 ) but after updating Hackrf  Xilinx CPLD , it was solved.

Now after reset from main power on button , capture app works well to me , does not keep hang up . 

It is true that the reset button , just freeze the screen , but maybe it is normal because it has its internal stand alone battery IC control , totally independent and that is a very bad issue!!!!  ... no way to recover any brick without dissasembling many screws  !  (Or at least I still did not find out yet  a user friendly way to recover it ) . When activating to DFU , that abnormal reset button does not leave to jump to Hackrf mode from DFU .

sdb333777, 2022-10-08:
update, so i followed these first steps from this article https://github.com/eried/portapack-mayhem/issues/651
dissassembly the set (separating both boards, reflashing Hackrf one to the latest Hackrf fw (that also updates automatically its Xilinx CPLD fw ) , and then reflash on top the Mayhem
now the capture does work from first startup and the jerky colorful meter spectrum analyzer is fixed as well. reset still does not work, just hangs/freezes the unit till power off

Brumi-2021, 2022-10-08:
Hi @sdb333777 , to avoid confusions , if I understood your post  well,  ,you followed my previous instructions , issue #651 ,
And you fixed both, issues : 
a-) now no jerky motion in the FFT waterfall
b.) no hang up in the Capture App , anymore .

Then to me , they are really very  good news ! !!!
Now both are in the same level.   (reset button is just  freeze the image / hang up , till new reset / power up) 

Regarding the reset button , it works well,  in the dissassembled boards , when using as independent Hackrf.
But once assembled with Portapack + battery ,  that "devil " has an isolated  battery managmenet  IC , and therefore the reset does not work properly , and to me created additional problems , when i need to unbrick it , (not user friendly as other models without battery . To  recover from DFU in that device , I need to dissassembly both boards each time . Not user friendly at all. Maybe there is another way to do it ,that I still does not know ...  )
cheers,

sdb333777, 2022-10-08:
yes, thank you. i was just so tired that i couldnt think or see right to read.

Brumi-2021, 2022-10-08:
Yes , I can imagine , 
 you have been lucky to meet me with exactly same device as you ... having all same  problems  that I already had with that "devil" ...
You make an accelerated trip in two days, that I took to me  months  to  figure out and solve it ....
Yesterday , solving  TX -  NO MODULATION , 
Today ,  jerky  motion  + hang ups  in the first Capture App  start up ...

Do not bring me more  new  problems , ...,  I do not have anything more in my pockets , ....
You should be really tired , with all those above issues ...
Cheers ,

GullCode, 2022-10-09:
Closing as the remaining problems can not be fixed (reset freeze).
BTW one solution is to add a switch ala H1 into you H2 so you can force cut the battery when needed without opening it.

closed 2022-10-09 by GullCode