portapack-mayhem/mayhem-firmware #724: i whan write custom firmware, but cant change image

girs1982, 2022-10-13:
where image and other files , i sucsec rebuild, and flash, but no changes , girst i want try change image and logo

eried, 2022-10-14:
You mean this: https://github.com/eried/portapack-mayhem/wiki/Create-a-custom-splash-screen ?

girs1982, 2022-10-14:
> You mean this: https://github.com/eried/portapack-mayhem/wiki/Create-a-custom-splash-screen ?

no it not all , i whant write my apps for starline keloq , how on my esp8266

girs1982, 2022-12-30:
just now i write starline decoder block for gnu radio https://www.youtube.com/watch?v=s1wNv0usPj4
and my next level must will app for portapack , but how write apps for portapack i dont know , and need manua and instructions.

Brumi-2021, 2022-12-30:
Hi @girs1982 , i saw your nice video , looks very interesting , but sorry , i do not know what is "startline" , is it a teleconf system ? 
Anyway , in the wiki , there are very good explanations about how to collaborate, creating your set up and the steps about how to compile and create your first PR .
https://github.com/eried/portapack-mayhem/wiki/How-to-collaborate

girs1982, 2022-12-30:
> Hi @girs1982 , i saw your nice video , looks very interesting , but sorry , i do not know what is "startline" , is it a teleconf system ? Anyway , in the wiki , there are very good explanations [...]

Hi,
Starline it car signal code , all source on my github

Brumi-2021, 2022-12-30:
I see , before I spelled wrongly it is "starline" car security control  (not "startline" , it is without "t" ) . Thanks

Brumi-2021, 2022-12-30:
Hi @girs1982 , I think that you already got the link explanations about how to collaborate , creating fw .
Then, let's close that issue   
We hope to get soon your new fw features  proposals!
Cheers

closed 2022-12-30 by Brumi-2021

girs1982, 2022-12-30:
> Hi @girs1982 , I think that you already got the link explanations about how to collaborate , creating fw . Then, let's close that issue We hope to get soon your new fw features proposals! Cheers

need more info , i cant find info about tx and transmit functions on portapack, in source all was not understanding

Brumi-2021, 2022-12-30:
Hi , it is true , that there are  a huge firmware files to learn, but step by step. 
Unfortunately we do not have any firmware manual book summarising all available fw  functions with usage examples...   
There is the wiki , in that repo growing day by day .
And additionally , as a landing , I think  it is worth to study Hackrf block diagramm and its schematic . I joined to this community  1,5 years ago and still there are many things to learn . But it is nice to re study all previous developper's job and try to improve it  dat by day . 
I recommend you ,first  to set up locally in your pc all arm  compile chain (as it is explained in the wiki) , and make small changes (changing simple things as labels ...) , and compile it and and create your local binary , and  try it reflashing it   And gradually increase the complexity or any new feature , and when you tested , validated , then release a PR to the community.  
The learning curve is slow , but definitely it worth while .
 Maybe @eried can add some more comment   .
Cheers

girs1982, 2022-12-31:
> Hi , it is true , that there are a huge firmware files to learn, but step by step. Unfortunately we do not have any firmware manual book summarising all available fw functions with usage examples... There is the wiki , in that repo growing day by day . And additionally , as a landing , I think it is worth to study Hackrf block diagramm and its schematic . I joined to this community 1,5 years ago and still there are many things to learn . But it is nice to re study all previous developper's job and try to improve it dat by day . I recommend you ,first to set up locally in your pc all arm compile chain (as it is explained in the wiki) , and make small changes (changing simple things as labels ...) , and compile it and and create your local binary , and try it reflashing it And gradually increase the complexity or any new feature , and when you tested , validated , then release a PR to the community.
> 
> If you can build some TX- Rx feature with gnu radio and hackrf , theoretically it should be posible to build it in mayhem , but unfortunately it is not as direct , and beside we have a limited cpu of 204Mhz aprox (not 2 or 3 Ghz of a laptop)
> 
> The learning curve is slow , but definitely it worth while .( Maybe @eried can add some more comment ) . Cheers

i was compile it, now i understand why flipper it not how portopack

eried, 2023-01-01:
:/ we don't have more detailed tutorials yet. I recommend you to check a similar app to what you want to achieve and clone that one as starting point. Maybe you can complete the wiki 😬