portapack-mayhem/mayhem-firmware #730: pas de son portapack h2 + (pcb 2014 clone hackrf one )

David642afk, 2022-10-23:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly verison here: https://github.com/eried/portapack-mayhem/releases)

----

bonjour, j'aimerai comprendre pourquoi il n'y a pas de son qui sort de la prise casque de portapack h2+,firmware utilisé 1.5.4      1.5.1      et nightly 2202-05-20, connecté au pc ces pareille sdrsharp, comment se fait il qu'avec tous les cerveaux qui sa save codé il n'y a rien qui fonctionne ,esque la raison ces par ce que se n'es pas du materiel original ,et que sa ne peut tous simplement fonctionné .en mode scanner rien pas une fréquence ,un talkie il capte ,mets hackrf seul ou avec portapack il n'y a rien .capture replay au pc conneté sa fonctionne avec urh bien sur si non faut oublié ,import de signal depuis urh en passant par audacity ,jai essayé de bidouillé sauf que ma pauvre sonnette elle à jamais sonné ,que du pc .je sais qu'il y a un moyen mets comment fait on ??? en tous cas sa donne mal au crane ,mets ces passionant .

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Tap on '....'
3. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release:
* Latest Nightly release:
* Previous working release:

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.

David642afk, 2022-10-24:
un truc bizzare aussi quand j'ai reçue le hackrf avec le portapack assemblé je précise ,la premiere fois qie je l'es allumé il n'y avait pas affiché en gros mayhem c'était écrand noir ,aprés j'appuyé sur le bouton ok et l"écrand s'allumé avec tous les menu du firmware ,aprés avoir cherché des infos ,il fallait appuyé sur le bouton gauche que sa initialise le lcd ,effectué et sa saffiché normalement aprés sauf que aprés avoir fait sa l'icone du son c'était barré je ne sais ou,2 mois aprés j'ai découvert le menu,usert interface ,hide H1 speaker options ,en décochant l'icone ré aparait,mets pas de son .ensuite j'ai éssayé de le faire fonctionné sans portapack connecté au pc rien non reconnu ni sur windows ni sur linux,il y avait rien d'installé dessus alors qu'il avait mits le portapack dessus ,faut'il les deux firmware installé ou juste celui du portyapack suffit il pour controlé l'ensemble ?.

gullradriel, 2023-03-29:
No description, no activity, only one to have a problem.
If you see this message and still have the problem, open a new issue with all the asked details filled.
Thanks.

closed 2023-03-29 by gullradriel