portapack-mayhem/mayhem-firmware #731: add noaa decoder

deboi238, 2022-10-24:
for the hackrf mayhem it would be nice if there is a support to decode noaa sattelites

Brumi-2021, 2022-10-29:
If you are talking of 137 MHz weather satellite band , I think that issue is duplicated to #121

BehleZebub, 2022-11-03:
I guess @deboi238 talks about NOAA weather satellite APT images @137MHz, APT encoding is explained [here (wikipedia)](https://en.wikipedia.org/wiki/Automatic_picture_transmission).
NOAA weather alert/SAME is something different, it's a terrestrial broadcast @162MHz and that's what #121 is about.

AFAIK [mamesshao8](https://github.com/jamesshao8/portapack-havoc-modified) has "RX NOAA" already implemented but the code is closed source.

I'd like to give it a try to implement a APT receiver, but it will take a long time to finish.

ImDroided, 2024-03-27:
Has any work been done on this? This would be a cool app. It sucks @jamesshao8 has decided to not contribute his work as he has done come cool stuff . I don't know who pissed him off but it wasn't worth it lol. He has noaa sat images decoder, weather fax decoder, color sstv rx and tx plus a bunch more.

dekid69, 2024-03-27:
Sadly there’s no work has been done on this, I really like this feature
like to be implemented to the mayhem

htotoo, 2024-03-27:
Can anyone provide a good quality c16 (pp recorder) recording? Not just the beginning (I only found that), but a bigger chunk of the image?

sergeyvfx, 2024-11-13:
@htotoo Hey! I have some I've been using for own decoder I've been working a while ago: https://github.com/sergeyvfx/radio_core/tree/main/data/test/apt

The [NOAA_44100.wav](https://github.com/sergeyvfx/radio_core/blob/main/data/test/apt/NOAA_44100.wav) is an audio recording from SDRSharp. I think it is 16bit int samples, 44100 samplerate. As a readme file suggests there seems some dropped packets in it. I never figured out what caused them, but seemed useful to have those to help debugging synchronization mechanism in algorithm.

There is [grid_test_44100.wav](https://github.com/sergeyvfx/radio_core/blob/main/data/test/apt/grid_test_44100.wav) which is generated.

Both WAV files have a decoded image next to them, so you can see what part they actually contain.

I do have a raw IQ capture of the transmission, but it is quite big. Not sure it is more useful than an audio file.

Not sure how far you're with your work on getting APT decoder. If it helps, the algorithm I came up with is implemented in  [decoder.h](https://github.com/sergeyvfx/radio_core/blob/main/source/radio_core/picture/apt/decoder.h), with a command line tool implemented in [decoder.cc](https://github.com/sergeyvfx/radio_core/blob/main/source/radio_core/picture/apt/tool/decoder.cc). It is quite naive implementation, nothing too fancy about it, but it worked quite good for the radio and antenna setup I've got. The tool is aimed for an offline usage, but the processor itself supports streaming.

There are also test files and decoders for SSTV, which might help with #2045.

Unfortunately, I am not sure whether I'll have enough time to dig into trying to implement the same in Mayhem, don't even have the portapack (yet :). But I am available to reply questions or even do more recordings! Just ping me if you think I can be of help.

I wish you all the best of everything! I really hope extra decoders can be implemented in Mayhem :)

htotoo, 2024-11-14:
I can't replay the wav from one pp to another, only the c16. And replaying is far more efficient testing, than waiting for the next pass.

BehleZebub, 2024-11-14:
@sergeyvfx thank you for the test files and decoder.
I don't know,  when I'll be able to look into this, but it surely helps for creating the NOAA app.

@htotoo it is possible to connect the output of one hackRF to the input of another, but connecting them directly will mostly fry the input stage, iirc. an -20dB attenuator should work fine to drop the max 15dBm output to the max allowed -5dBm of the input circuit.

sergeyvfx, 2024-11-14:
@htotoo There are a few things I can think of:
- It should be possible to feed audio wav to a PortaPack, and let it transmit it as NFM/WFM (with proper deviation), and pick it up by another PortaPack. Can use some coax and attenuator to avoid extra noise and distortions introduced to the system. Essentially, convert PortaPack to a signal gen with modulation support.
- Maybe even PortaPack supports feeding audio as a file for transmitting? Something similar to how it transmits APRS, with the difference being the audio wouldn't be generated but read from a file?
- If I am correct, c16 is a 16 bit signed PCM, raw. So to convert wav to c16 you can either use some Python script or Audacity using RAW format? Since the files I have online are audio, maybe Python script is a better choice, as then you can take care of modulating the signal into a carrier.
- Can use GNU Radio to take care of modulation. And then either write the signal to a File sink (not sure it supports c16 directly, but it should be relatively possible to convert it to a proper format), or, feed to a HackRF from one of the PortaPacks.

@BehleZebub Understandable, no worries!

htotoo, 2024-11-14:
Noone wanted to wire 2 pp-s together, they have antennas ;)
Also would be preferable to have a big c16, that has a lot of other things too, not just a selected. that could help with the filtering, ...

So when playing a decodeabla signal from one pp to another, that has "everything" in it, is a more likely real world usage, than the wav playback. And i would love to prepare to a real world usage, not a "lab one".

[image]

sergeyvfx, 2024-11-14:
The `NOAA_44100.wav` is an actual recording, just of audio, not an IQ. If you know a good place where i can share 190 meg of file I can provide the IQ recording of the same pass.

There is no single real world scenario, everyone has different antennas, different weather conditions, etc. At the end of a day what matters is SNR. In the "lab" configuration you can simulate different SNR levels, and say "the decoder will lock onto a signal once you have N B of SNR".

It is also way easier to start with the cleanest input possible, and put the foundations of the decoder, and make sure those work. Dealing with the more noisy scenarios is more like tweaking tolerances, filters, etc. Starting with a mindset to write a decoder that handles the lowest possible SNR is quite hard.

Brumi-2021, 2025-03-25:
Hi @sergeyvfx , nice to meet you .
I also have some old recorded audio files from NOAA 137 Mhz , but i will download yours aswell. 

Soon @htotoo and my self will start to develop this satellite map  Mayhem App decoding . But it will be in black and white , not as nice as the famous wxtoimag or the SatDump... 

But it would useful if you can also share with us  some raw IQ recorded file  pass . 

Lets keep in touch 
Cheers

Brumi-2021, 2025-03-25:
Thanks for sharing those samples. 
I also have some old ones ,about demodulated audio . 
I will check and investigate using both , yours and mine ,  thanks 

[image]

htotoo, 2025-03-26:
Also needs to note, that there won't be map overlay on our solution as it would need a LTE database updated regulary, an exact GPS location, and the right time on PP. In theory it would be possible, but the error factors would be huge.

Brumi-2021, 2025-05-10:
Hi , dear all, @sergeyvfx , @ImDroided . @dekid69 , @BehleZebub ,  be ready to start the LIVE  CAPTURE TEST, 
Soon we will have a new stand alone NOAA App, #2648 ,  able to record directly in a .bmp file the NOAA weather satellite map . 

I used @sergeyvfx captured files to develop and debug together with some of my old captures . 
There are some pending fine tunings and optimization jobs , but the basic structure and functionality is there . 

Here you can see some captures from my test replay files,  (including some test patterns from @sergeyvfx  , thanks :-)  ) 
[image]

[image]

Be ready to track the 3 x NOAA satellites and let's tune the 137 Mhz band . 
Not sure if we will need a big QFH antennas or just with some dipole + preamplifer will be enough ... Let's see ...
As soon as merged that commit , we will close that issue , and let's start the FIELD TEST , and pls. share your captured images through discord . (we will need to create a separated discond NOAA channel ) . 

Cheers,

closed 2025-05-13 by htotoo

ImDroided, 2025-05-13:
I have a qhf on my roof with the sawburn amp. I will try to test this
tonight after I get home from work on the next satellite pass.

Brumi-2021, 2025-05-31:
Hi , dear all, @sergeyvfx , @ImDroided . @dekid69 , @BehleZebub , 
did you try to capture any NOAA  satellite pass ? Please could you share it in discord, # NOAA section ? 
I think still possible to improve several dB's more sensitivity and we may add , the framA , frame B sync selection. 
Thanks

sergeyvfx, 2025-05-31:
Hey everyone, great progress, thanks for working on it!

@Brumi-2021 Unfortunately, no, had limited time for hobbies in the past weeks.
Would you need a pass recorded by the portapack, or any other radio will suffice? (I still didn't get across of getting the portapack for myself)

ImDroided, 2025-05-31:
I haven't had much time for hobbies either lately. I'll try to update my
Porta pack this weekend and see if I can capture a pass on it.

Brumi-2021, 2025-05-31:
Hi to both , @sergeyvfx and @ImDroided ,  thanks to both for answer us . 
No problem, I just wanted to know,  if you knew about it .
So far we have two ways of capturing : 
1-) by direct NOAA App --> you can see the incoming through LCD, hear it ,  and record  picture.bmp ,  

2-) or  by Audio App , selecting Audio App , WFM AM NOAA mode selecting 38khz , and see waterfall , and hear , and record  APT_signal.wav audio file . ( I think the first one , has some local interfence beat,  the second one is cleaner , but it needs external tool to extract from APT_signal ---> picture.bmp. 
 
Take your time , no need to rush .  
Based on my test,  so far , possible to capture with QFH antenna (and better if we have some pre-amplifier with 137 Mhz saw filter).   
But I wonder if some user also tested with just dipole . 
Cheers.