portapack-mayhem/mayhem-firmware #732: OP25 Trunking Scanner? Is it possible?

maietta, 2022-10-29:
So I've been thinking about this for a few days now and realized I have my Portback H2 and hackRF. Obviously, the radio is quite capable and the software I've seen is written in C.

I'd like to know if it's technically possible to listen to a single frequency while also listening to another. I've never written any code for the hackRF and think this could be a good project to get familiar with the internals a bit more.

Does anyone have any knowledge of the limitations of listening to multiple frequencies at the same time, or more specifically, always listening to a control group frequency while simultaneously listening to other frequencies based on the data found in the control frequency? Some decoding would have to be done.

If this is a rediculous idea, great. If it's possible, then I would like to move forward with spending some time trying to code this.

sdb333777, 2022-10-29:
do you have a usb rtl-sdr ?

maietta, 2022-10-29:
@sdb333777 I have a pair of nano 3's arriving any day now. They are based on the R820T2 tuner and use the  RTL2832U IC for demodulation and USB interfacing.

sdb333777, 2022-10-29:
dsd+ does trunking very well. or even better dsd+fastlane
fastlane works with 1 rtl-sdr, but if you wanted to use the hackrf one you could use sdr# and then pipe it to dsd+

maietta, 2022-10-29:
That could be useful for my Broadcastify feed source setup I'm working on using both SDRs, which is what I bought those for. My question though is about the possibility of going portable with just the hackRF and Portapack combo. I think a P25 Phase 1 and 2 trunking scanner feature would be a great addition to the Mayhem firmware.

sdb333777, 2022-10-29:
i asked them about that, they are limited on space and the portapack doesn't have the speed to do the decoding

maietta, 2022-10-29:
Thank you. While that's too bad, it's a good thing because I have enough projects to work on already. Thank you for such a quick resolution on this.

closed 2022-10-29 by maietta