portapack-mayhem/mayhem-firmware #733: ADS-B squawk always 0000 in FW 1.5.4

MerApple, 2022-10-29:
Hello
I am experimenting with ADS-B transmission, I am transmitting from portapack H2 with the offical stable release 1.5.4 I have selected to transmitt squawk from the menu, I am recieving the transmission with SDRangel (i have confirmed that reception and decoding ADS-b works in SDRangel), when transmitting from portapack I can see in SDRangel that squawk is 0000 regardless of what number I set the squawk to from portapack.
Is this per design? Maybe changing squawk is not allowed?

Brumi-2021, 2022-12-06:
Hi @MerApple   , I  tryied to check about your issue , 
and I could confirm two points : 

(1) I can confirm, exactly what you mentioned  : testing with SDRANGEL  version 7.81 as a receiver  , I could not receive any ADSB-SQUAWK code from our Portapack-mayhem, fw  n_221121 from  ADSB TX  App.  (most of other parameters yes ) 
(pending to see with external regular flights if that  squawk code is well decoded in Sdrangel ,  or always also stays blank ??? (I still did not check it , how about you ? ) 

(2) I also could see , that when I input in Hackrf ADSB TX app ,  BEARING  to 
        10º --> it is displayed in SDRANGEL as  80º (complement to 90º)
        20º --> it is displayed in SDRANGEL as  70º (complement to 90º)
        (not sure if  the mistake is in mayhem side or  in sdrangel side,  t.b.c.  )

[image]

Now I still have doubt about where is the problem , in  TX side (mayhem fw ) or  in RX side (sdrangel) , 
But as sdrangel has a much intensive debugging and users,  most likely it will be in our side .
Cheers,

ImDroided, 2022-12-08:
You can test this further and switch out your ADS-B receive software and see if it persists, if it does then the issue is on our side. If it is on our side then you could flash a older release of Mayhem to see if it works there. Post your findings.

Brumi-2021, 2023-03-03:
Hi @MerApple 
This issue ,  could be closed , once merged the PR #818 .

Waiting for your feedback !
In my test ,   (dump1090 always capture it ,  sdrangel ,not always ,..., both with RTL-SDR)

Cheers,

gullradriel, 2023-03-04:
Closing as #818  was merged, thanks @Brumi-2021  :-)

closed 2023-03-04 by gullradriel