portapack-mayhem/mayhem-firmware #735: portapack mayhem unfortunately my pc

zefaxx, 2022-10-29:
----
hello I currently have the portapack mayhem unfortunately my pc does not detect it could tell me why?

(Please try the latest nightly release before submitting this. You can find the latest nightly verison here: https://github.com/eried/portapack-mayhem/releases)

----

**Describe the bug**
A clear and concise description of what the bug is. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Tap on '....'
3. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release:
* Latest Nightly release:
* Previous working release:

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.

eried, 2022-10-31:
is it in hackrf mode?

alma3lol, 2022-10-31:
Had the same issue. Just install drivers from 'driver' folder
[image]
[image]

gullradriel, 2023-03-27:
No answer from OP. Should have been driver related as it's the only issue talking about it.