portapack-mayhem/mayhem-firmware #742: 2 request for portapack

alexxmarsh44, 2022-11-05:
1. Is there anyway you could implement a direction finding app to go with the operacake add on board for the hackrf? It would be cool if it could run the operacake and provide a realtime animated compass that points to the direction of the RF signal in question. [https://www.rtl-sdr.com/hackrf-opera-cake-released-a-rapid-rf-switching-board/](https://github.com/eried/portapack-mayhem/issues/url)

2. A spectrum analyzer that makes use of the hackrf_sweep function and displays a set section of the spectrum on a PSD plot and possibly a waterfall.

GullCode, 2022-11-08:
One have to buy an operacake to each of the developers so we can try our best X-D

eried, 2022-11-09:
2. is a bit covered already by https://github.com/eried/portapack-mayhem/wiki/Looking-Glass 

I think there is not much extra capability that can be added due resolution/hardware

gullradriel, 2023-05-10:
We do not have opera cakes under our hands nor we know it we can talk to it while in portapack mode.
1. partially covered as there are bew apps helping for direction finding
2. covered with latest app update

closed 2023-05-10 by gullradriel