portapack-mayhem/mayhem-firmware #743: Radiosonde meteomodem M20 decoding

przemo3679, 2022-11-06:
There is new type of radiosonde from meteomodem: M20.

Unfortunetly I know nothing about decoding them.
At current state mayhem 1.5.4 recognise it as a : meteomodem ??? and decodes only DateTime.

It would be nice to be able to decode at least position coordinates from the signal.
This would help at sonde hunting proces.

GullCode, 2022-11-08:
Maybe you can try to look at how it's done and adjust a new setting ?

przemo3679, 2022-11-08:
Well, i'm not a programmer unfortunetly.
But there is this project on Github: [radiosonde_auto_rx](https://github.com/projecthorus/radiosonde_auto_rx).
It is open source, and found there how it decodes M20.
As I said, I'm not a programmer, but it looks vewry similar to M10 which is alredy coded.
Probably code needs just a few tweaks.

GullCode, 2022-11-08:
I'm telling myself we should try to reach the author and connect him with someone familiar with how portapack is doing the actual work for his decoding, and try to add some more with their help.

joyel24, 2023-04-28:
> Unfortunetly I know nothing about decoding them. At current state mayhem 1.5.4 recognise it as a : meteomodem ??? and decodes only DateTime.

After reading a bit the code I can say that the DateTime is not extracted from the rf packet, so not decoded. I've not tested but if you change the date of your portapack, I think you will see the change. ;)

I made my first tests yesterday to try to support meteomodem M20 but I had to wait for a new launch and climb a mountain to test the result as I did not git any sonde before.

But fortunately the sky gave me one at less than 15km today so it will be easier now !

przemo3679, 2023-04-28:
> I made my first tests yesterday to try to support meteomodem M20 but I had to wait for a new launch and climb a mountain to test the result as I did not git any sonde before.
> 
> But fortunately the sky gave me one at less than 15km today so it will be easier now !

Unfortunetly i can't help you with coding, but i have several M20, device that decode them correctly (TTBeam + MySondyGo), and HackRF with Portapack.
If You need help with testing just hit me up, so we can work together.

joyel24, 2023-04-28:
> Unfortunetly i can't help you with coding, but i have several M20, device that decode them correctly (TTBeam + MySondyGo), and HackRF with Portapack. If You need help with testing just hit me up, so [...]

I don't know how you can help me afar but If you wish to send me a TTGO it will be very helpful for me to take new ones as I also plan to try making a new firmware for the stm32 chip running on this M20 radiosonde to use it in other projects. 
I only have one for the moment so I don't want to brick it...
😊

joyel24, 2023-04-29:
DONE ! 😁
#932

joyel24, 2023-04-30:
This issue can be closed...

closed 2023-04-30 by gullradriel

joyel24, 2023-04-30:
> Unfortunetly i can't help you with coding, but i have several M20, device that decode them correctly (TTBeam + MySondyGo), and HackRF with Portapack. If You need help with testing just hit me up, so [...]

Finally I have a job for you if you are free now...
Flash this firmware on your portapack https://github.com/joyel24/portapack-mayhem/releases/tag/a74d0e5
Power on an M20
Take your TTGO and compare the altitude with portapack.

Thank you

przemo3679, 2023-04-30:
Rly sorry mate, 29.04-3.05 are holidays in my country, and im out of home. As soon as i get back i will flash and test your relase. Thanks for patience.

przemo3679, 2023-05-02:
Well, it seems to work to some extend. 
I've had M20 fly-by just now and this is the resaults.
Note: difference between alt and coordinates are caused by delay between pictures. As is now it seems pretty accurate
Thanks for your hard work mate.
[image]
[image]

joyel24, 2023-05-03:
Really nice ! Thanks for your report. 🙂
Here, the wind direction changed and I have not been able to capture any packet since my recent work on the altitude decoding.