portapack-mayhem/mayhem-firmware #746: Hackrf Mode Black screen

JoseAbreu28, 2022-11-11:
My PortaPack shows a black screen when I put it in hackRf Mode, I can't connect it to the computer and some features also crash

Brumi-2021, 2022-11-11:
Hi , I guess that till now , always your HackRF mode was showing a blue screen, not ? 

If that is the case , follow below steps ,
maybe in your last flashing process, you got some random short periode or interruption of the good USB communication l (It could be bad USB quality cable or bad USB connector pin contacts or just a moving or some impact while flashing ). Anyway do not worry , you have to put your device in DFU mode and the follow that DFU process , https://github.com/eried/portapack-mayhem/wiki/Update-firmware-troubleshooting

  Basically , 
1-) connect USB to the PC and Hackrf . 
2-) put your device in DFU mode (easy check wiki)
To invoke DFU mode: Press and hold the DFU button. While holding the DFU button, reset the HackRF One either by pressing and releasing the RESET button or by powering on the HackRF One. Release the DFU button.
3-) send command from a proper folder where you have the files , like the link (to unbrick) 
dfu-util --device 1fc9:000c --download hackrf_one_usb.dfu --reset
That should recover your normal Hackrf mode including blue screen
4-) send normal flash command with the fw version you want . (As you already know ) 
Reset and it should be fully recovered . 
cheers

jLynx, 2022-11-18:
Have you tried these steps already? https://github.com/eried/portapack-mayhem/wiki/Won't-boot

jLynx, 2022-11-18:
This is a known bug. Please ignore it for now. Everything will work fine, it's just the screen that won't show anything

jLynx, 2022-11-18:
Reopening this issue as I can't quite find the duplicate ticket at the moment

bhavsec, 2022-11-23:
Flashed my PortaPack H2 from China with latest stable mayhem v1.5.4 release. Everything is working perfect except HackRF mode. After clicking "Yes" It goes to black screen (Empty Screen with no symbols) also with no USB light ON when plugged into PC. How can i use HackRF without removing all screws and portapack board?

Brumi-2021, 2022-11-23:
Hi @bhavsec , are you working with linux?
Based on your symptoms, 100% that you have a Portapack with big CPLD QFP100.  
But it should be also ok . If you have all 1.5 4 with full functionality, you should have a correct Hackrf mode but with black screen . Then in that state , pls plug usb to your pc till getting stable  green USB light on  . From that point it should be full functional the Hackrf mode , but just with black screen. That it is . You can reflash or control by GNURADIO....

(Note it is possible to prepare manually  special 1.5 4 bis one shot binary with blue screen in Hackrf mode , but we are not doing because it needs split Hackrf fw branch from official  upstream, and because I have some other compiler compatibility problems with current fw and SD).

bhavsec, 2022-11-23:
Thanks @Brumi-2021 for the reply, I am using MacOS. I bought PortaPack from OpenSourceSDR Lab, I have performed the steps as you suggested, plugged in and out 5 times still there is no luck. USB Light is not turning on, LCD Display remains black with no output. 

After checking documents from my vendor, they have replaced CPLD 5M40ZE64CN5 by a new EPM240T100C5N. The current Github Mayhem firmware checks for 5M40ZE64CN5 during initialization and the hackrf can not be recognized by my computer when I input command “hackrf_info”, and the USB Led of hackrf is not on. 

They also mentioned in order to use HackRF I have to go to dfu mode and flash hackrf original firmare. I don't want to do this step everytime I want to go to HackRF mode and switch between mayhem, is there any way we can fix this issue?

bhavsec, 2022-11-23:
I was doing wrong all the time, I was inserting USB Cable after going to HackRF Mode, But I have to do insert it before, 

I just found the way for it to work:

Step 1: Turn Off Hackrf+Portapack
Step 2: While it's OFF plug the cable to Hackrf+Portapack and computer.
Step 3: After plugging in, The device will turn on automatically, go to HackRF Mode and press "YES"
Step 4: Type `hackrf_info` and it'll Connect Successfully.

Brumi-2021, 2022-11-23:
Hi @bhavsec , I am glad  to see that you found a way to go to Hackrf mode .
It is really strange , I also have one Portapack with EPM240T100C5N CPLD , and in my case , I got black screen in Hackrf mode , but with  green LED on , when having the USB link active .  

But happy to see that you have a workaround solution.
Thanks for sharing it,
Cheers

closed 2023-03-27 by gullradriel