portapack-mayhem/mayhem-firmware #751: GPS-sim: Receiving only one satellite signal on Android

HaXlezi, 2022-11-21:
Hello guys,
I was trying to spoof location on Samsung Android device using GPS-sim. I think I generated file correctly using manual on [https://github.com/eried/portapack-mayhem/wiki/GPS-Sim](url). After starting the simulation my phone was receiving signal from **only one satellite** which is not enough to get a fix obviously (shown on picture). It is weird because it looks like the SDR transmit some signal but only for one satellite. Changing of transmitting power didn't change anything. I tested 3 Android phones. Airplane mode was on.
Do you have any ideas?
Thanks a lot!

[image]

Brumi-2021, 2022-11-21:
Hello @HaXlezi , let me add several comments .
I-) when I checked on PR #487 , I could get many satellites , but 20 secs is not enough , and as @beged mentioned , the accuracy of the TXCO is very important .

ii-)  if you did not -pls  also the wiki GPS-sim part ) and eventually we can also consult it to @beged , he is much more specialist about GPS.

Cheers

eried, 2022-11-22:
If the three phones see the same, then I agree with @Brumi-2021 and there is something about the hackrf. Have you tried with https://github.com/osqzss/gps-sdr-sim ?

HaXlezi, 2022-11-24:
> Hello @HaXlezi , let me add several comments . I-) when I checked on PR #487 , I could get many satellites , but 20 secs may not be enough in some mobiles to lock, and as @beged mentioned , the accuracy of the TXCO is also very important .
> 
> ii-) if you did not -pls also the wiki GPS-sim part ) and eventually we can also consult it to @beged (https://github.com/beged) he is much more specialist about GPS.
> 
> Cheers

Hi Brumi, thank you for response!

1) I think the time is not the problem. Usually I was waiting 1-2 minutes to get a fix before I moved to another gain number. And whole the time there was only 1 (exceptionally 2) satellites. About TXCO - I'm not sure what this component is doing, can you explain please? And how can I regulate its accuracy? Is TXCO included in HackRF + Portapack as it was bought from Aliexpress?

2) I checked the wiki before. It was my manual, how to do the whole thing :)

Thank you.

HaXlezi, 2022-11-24:
> If the three phones see the same, then I agree with @Brumi-2021 and there is something about the hackrf. Have you tried with https://github.com/osqzss/gps-sdr-sim ?

Hi eried, thank you for response!
Yes. I build the app for generating c8 files in Visual Studio corresponding to gps-sdr-sim project

Brumi-2021, 2022-11-24:
Hi @HaXlezi ,
If you created the manual , you probably have more know how than me about all those issues, 
Anyway , let me share some relevant issues :
First  , let me copy the comments that I got  from @beged ,when i was testing GPS Sim App, and we were talking about the locking time  : _"yes the 20 sec records may not be enough... depends on phone too... but usually longer is required, and there are several other factors, like you need a TCXO ref GPSDO ref externally injected for very precise frequency and so on..."_

Just to share with you , recently most of the Portapack usually comes with TCXO (Temperature Compensated Crystall Oscilator 
(but in some models you could add it , with some plug  add small boards)
[image]
[image]

As you could see  in this  chart ,  the TCXO  improves significantly compared to a normal crystal ,  the frequency accuracy , reducing its  possible freq. Hz  deviation a lot  (from the central nominal one) . 
[image]

And if you check that link,  http://gpsinformation.net/main/gpslock.htm,  you can see in that paragraph , 
"The second dimension is frequency. The receiver must correct for inaccuracies in the apparent doppler frequency of the satellite and "zero-beat" the signal. For reasons that I won't go into here (related to the 1 msec code repeat cycle and a fellow named Nyquist), the coarsest frequency steps **_a receiver can take during initial acquisition is 500 Hz,
but the receiver's crystal oscillator may be off by up to +/- 10 kHz or so, so there are 20ish frequency cells that have to be tested._** And this process must happen for all N satellites that are in use. So, the acquisition process can be quite laborious." 
==>   Then I understand , that if our receiver has  a longer  shift freq. deviation , it  will take longer time to lock .

Cheers,

eried, 2022-11-25:
> > If the three phones see the same, then I agree with @Brumi-2021 and there is something about the hackrf. Have you tried with https://github.com/osqzss/gps-sdr-sim ?
> 
> Hi eried, thank you for response! Yes. I build the app for generating c8 files in Visual Studio corresponding to gps-sdr-sim project

but you can also transmit them with this, just for the sake of testing the hackrf independently to the sd+portapack :)

gullradriel, 2023-05-10:
Closed for inactivity

closed 2023-05-10 by gullradriel