portapack-mayhem/mayhem-firmware #753: TransmitterModel::disable() does not wait for TX to complete

lujji, 2022-12-05:
BasebandThread passes a 2048 bytes dma buffer to the corresponding BasebandProcessor. When last chunk of data is written into the buffer, BasebandProcessor instance will signal back to the app with a `txprogress_message.done`.  The app will then call `transmitter_model.disable()` to stop the transmission. This, however, will not guarantee that all samples have been transmitted, and will result in incomplete transmission.

This issues can be easily reproduced by filling the whole 2048 buffer with fixed number of samples and stopping the transmission.

gullradriel, 2023-04-28:
@lujji did you think about a solution ?

lujji, 2023-07-12:
@gullradriel the solution would be to poll for transfer complete signal from DMA peripheral before disabling transmitter. Unfortunately, I didn't have enough time to get familiar with the LPC part to create a PR. As a quick work-around a short delay can be added before calling disable().