portapack-mayhem/mayhem-firmware #756: The ability to decode FM HD radio also known as (NRSC-5) digital radio.

fierytechknowhows, 2022-12-13:
**Describe the solution you'd like**
I'd like the ability to decode something called FM HD digital channels also known as NRSC-5. The FM digital channels are broadcasted alongside the main analog FM radio station. To find them Just look at an FM radio station in the US on a spectrum analyzer and you'll see sidebars above and below the center analog frequency. I'd like an option in the Receive menu with the ability to decode the audio, display the title of the station, and show the music titles and artists in the Mayhem firmware.

fierytechknowhows, 2022-12-19:
I'm guessing no one else knew those sidebars on US FM stations were other digital channels that play different music provided by the same station. I didn't know until recently myself and had always thought of those sidebars as either stereo left and right analog audio channels or start and end frequency bars marking the start and end of the analog FM radio station.

Brumi-2021, 2022-12-19:
Hi I did not know  about that USA FM digital encoding .
Anyway , here seems to be  a nice introduction link including  Argilo’s  GitHub code. 
https://www.rtl-sdr.com/decoding-and-listening-to-hd-radio-nrsc-5-with-an-rtl-sdr/

DaveNF2G, 2026-10-07:
Add my vote for this one, too.