portapack-mayhem/mayhem-firmware #759: Rolling code Catcher

Xtt5, 2022-12-18:
Please add a Rooling Code Catcher to the menu

eried, 2022-12-18:
what is "Rooling Code Catcher"? can you explain what functionality do you expect?

Xtt5, 2022-12-18:
It recognizes a signal and jams it so that it does not arrive at the point of origin in order to bypass the rolling code

Xtt5, 2022-12-18:
I just wish there was a way to bypass rolling code
thanks for your answer :)

eried, 2022-12-19:
There is a jammer app, you can set the frequency of the keyfob there

somethin9, 2022-12-19:
wanna steal something?

Xtt5, 2022-12-20:
No, I have a few things with rolling code myself and I am very interested in such security gaps. If I wanted to steal something, I wouldn't write it here

Einstein2150, 2022-12-22:
> Yes, I know, but I mean an app with which you can jam and record frequencies at the same time

HackRF is a half-duplex transceiver. This means that it can transmit or receive but not both at the same time.

eried, 2022-12-22:
Yeah, I don't think this request is doable. I will close this for now.

closed 2022-12-22 by eried