portapack-mayhem/mayhem-firmware #761: Largura de banda

samukas81, 2022-12-19:
Adicionar largura de banda de 100khz na função espectro. Obrigado.

eried, 2022-12-19:
I guess you want more granularity? in which part?

Brumi-2021, 2022-12-19:
Hi @samukas81 , regarding your request to add more freq. decimals to the selected freq. , I reduced the steps in that PR #710 . You can easily select it in the Audio RX App ., reducing the steps to 50 and 10 Hz , but we agreed to not increase more decimals in the UI ,  to save the UI space . Is it not enough ?

samukas81, 2022-12-21:
pode ao menos adicionar largura banda 100khz em SPEC?

Brumi-2021, 2022-12-30:
Hi @samukas81 , I do not understand you .
Current SPEC mode  has already a config selectable BW field ,  from 500khz to 20 Mhz   .  Then I do not understand, when you request  us to increase the BW from 100 Khz in SPEC mode ?

samukas81, 2023-03-27:
This modification would be very useful!!

gullradriel, 2023-04-07:
Added, will be available in next tomorrow.

closed 2023-04-07 by gullradriel

samukas81, 2023-04-07:
obrigado!

samukas81, 2023-04-07:
I don't know the right place to post this but it's just a tip.  The component responsible for the amplification of the TX (mga81563) analyzing in the datasheet shows that the minimum operating frequency is 100Mhz, I believe to be the reason for its failure.  I believe that using TX above 100Mhz does not happen what happened to me.  Sorry if this isn't the right place to post, if you could add it somewhere on the homepage that would be nice.
[image]