portapack-mayhem/mayhem-firmware #762: OOK scanner question.

Commander-Crash, 2022-12-20:
I just got the latest nightly build and no i am not reporting a problem, however i do have a question and it may be a n00b question i do apologize in advance.
I notice that the bit length is us (microseconds) is also sort and long at the same time. Will this work? Also is 24bits the max that the portapack can do or is some work still being done. Thank you for the response in advance.

lujji, 2022-12-20:
There is fine and coarse adjustment for bit length. What you see on the screen is the actual bit length. You can increase MAX_DE_BRUIJN_ORDER in proc_ook.hpp, but I haven't tested with larger values, since scan takes unreasonable amount of time.

gullradriel, 2022-12-21:
question answered.

closed 2022-12-21 by gullradriel