portapack-mayhem/mayhem-firmware #764: Please add the snake game

Xtt5, 2022-12-20:


gullradriel, 2022-12-21:
Adding those kind of thing isn't planned. Maybe you can do it yourself and make it a PR.

closed 2022-12-21 by gullradriel

NotherNgineer, 2023-05-14:
I too was thinking of adding some simple games, if there's enough ROM/RAM space.  I'll have to code one up.  :-)

gullradriel, 2023-05-14:
There are only 1024 k in the flash.
Maybe make a dedicated firmware which aim will be to make a game boy with the hackrf ?
I'm slightly kidding, because it's one of the best solution. You can side load a lightweight version of a bin.
But inside the Mayhem firmware, it's still a big no for me.

NotherNgineer, 2023-05-14:
Perhaps a HAM Radio licensing exam practice "game" would be relevant for the amateur radio folks (questions would reside in a text or JSON file).

But I see we're uncomfortably close to 1024K already!