portapack-mayhem/mayhem-firmware #770: File Manager (inside Utilities) won't open some SD cards 16GB

Brumi-2021, 2022-12-29:
----
Problem description : 
Recently , I can operate with my SD card 16GB scan Dish Ultra ,micro SD HC I , 
with Capture and Replay without any problems , but I can NOT  open  File Manager App (inside utilities) . 

If I go to Debug --> SD Card , I can see the following screen , 
[image]

Additional information , 
if I input another, very old one  16GB Transcend SD card , Mcro SD HC , the File Manager app (inside utilitities ) it works well . 
[image]

I confirmed the same behaviour  in many   fw version 1.6.0 ,  1.5.4 , 1.4.3,  1.31 , 
 it seems more related to the SD itself .   (fyi , both SD's  can be  opened  well  by linux PC ). 

Then the File Manager app it seems working well , because when I input another old one 16GB from another device , it can be seen its contents .  But  not  the one used in the Portapack .

I wondered if File Manager App can NOT see any contents  , with extended format  using 16GB or more ???( > maximum  4GB format ????? ) but if that is the case, why I can see all recorded captured C16  files from Replay App , or play all  the GPS Simulate files ,  or  play the Soundboard wav files , or the SSTV files , or the ADSB map, or take screen pictures without any problem ....?  
(so the access of all files works well R/W , but the File Manager can not be opened in the same SD , any clue ???) .

thanks

dominch, 2023-01-03:
I just struggled to get 32GB card working and at the end I needed to use only first 16GB
Try to do the same thing - for larger card use 16GB partition and format it with fat32 (4k cluster), that work for sure

zxkmm, 2023-04-26:
Can you test if it's related to PR#905, PR#906 aka Refactor? if so, i can figure out way or i'm ok if you decided to revert it.

gullradriel, 2023-04-26:
Unrelated don't worry.
I just made a cleaning tour around the bugs hence resurrecting the activity of some :-)

Brumi-2021, 2023-05-01:
Hi @gullradriel and @zxkmm , I think big probability that it is exactly same as #916 , then let me close it .
Cheers

closed 2023-05-01 by Brumi-2021