portapack-mayhem/mayhem-firmware #773: Problem upgrading the firmware

ZaneS25, 2022-12-29:
**The issue**
When I am trying to upload my Portapack H2 and it will not let me update, I keep getting an error code when I'm in the computers  CLI (when you click on flash_portapack_mayhem.bat) saying that it is not found. I have tried DFU mode and have tried several other versions of upgrading and nothing has worked, I've even tried using several different softwares, windows and Mac.

**What I Did**
1. Load it all up like normal (unzipping, and opening)
2. Put the Portapack into HackRF Mode
3. Plug it in and open CLI and tap a Key (I have tried plugging the Portapack in even when the cli is open, every variation)

**My Hardware**
I'm using the Portapack H2, Version: 1.4.0 running the "Mayhem firmware"

Im trying to update to the newest stable Firmware available, When I purchased the Portapack it came pre installed with ver. 1.4.0

I would love any help to fix this issue and what is wrong, Thank you!

tjjdesign, 2023-02-06:
Same here.
[image]

jLynx, 2023-02-06:
Can you try boot into dfu mode and flash it with the dfu file?

fiveseven808, 2023-02-08:
I have a similar issue. I have no idea how to check what version of mayhem my portapack h2+ is already running though. 

I can't use the HackRF mode flash method, however HackRF mode launching from the portapack kinda works (SDR "works" in SDR# but the noise doesn't make any sense, i.e. I can't even listen to WFM radio stations)

With my portpack removed, I can put my hackrf board into DFU mode, and use the dfu_hackrf_one.bat file, which just throws it back into HackRF mode which doesn't help me. 

What I'm trying to say is that the DFU mode seems to put the hackrf into hackrf/usb mode. The portapack can do the same from the controls/screen. In both instances, the rest of the batch files do not seem to recognize my portapack/hackrf and refuse to flash it. 

Even though the batch files refuse to flash it, I can still launch SDR# and get some response out of the hackrf (i.e. if I hold onto the antenna, the signals in the FFT rise. 

I'm at a loss at what to do. I even tried following some instructions here 
https://github.com/greatscottgadgets/hackrf/issues/456

MattPayne76, 2023-04-25:
Hi, I had the same problem and solved it by installing the dedicated driver provided in the zip file, try it if you didn't already, maybe you have the same issue.

Also, if you have black screen at power up after a firmware update, turn the device off and hold the left button while powering up (only the first time).

MattPayne76, 2023-04-26:
> > Hi, I had the same problem and solved it by installing the dedicated driver provided in the zip file, try it if you didn't already, maybe you have the same issue.
> > Also, if you have black screen at power up after a firmware update, turn the device off and hold the left button while powering up (only the first time).
> 
> Sorry off topic, but @MattPayne76 are you from NZ? I happen to know someone with the exact same name

Hi jLynx, sorry for disturbing so ! I am from France actually, Payne is just because I liked the game Max Payne :)

gullradriel, 2023-05-10:
Closed due to inactivity

closed 2023-05-10 by gullradriel