portapack-mayhem/mayhem-firmware #776: Issue saving freq. in audio and unable to create new catagories

grinchdubs, 2022-12-30:
**Describe the bug**
I'm no longer able to save frequencies In 1.6.0 in the audio app and new categories in the freqman app.
Upon further investigation, when I try to edit the txt file in the freqman folder in a text editor, I am no longer able to see the list on the pp going back to the nightly from 12-18-2022. The list is empty even though it shows a file size in the freqman app. and data is in the list.

I had to revert to 1.5.4 to be able to create, add and edit a custom freq list
I was then able to edit the file with a text editor and then see on pp
and yeah back on 1.6.0 and same result

**To Reproduce**
Steps to reproduce the behavior:
1. Go to audio app
2. Tap on freq.
3. Save freq
4 Load freq

**Expected behavior**
I expect to see the newly saved freq in the saved category however it isnt found and the category is empty. Happens even with a new category, however that is sometimes not seen when created.

**Affected versions**
* Latest Stable release: 1.6.0
* Latest Nightly release:  Nightlies going back to 12-18-2022
* Previous working release: 1.5.4

**Additional**
After going back to 1.5.4, I deleted the category and recreated it in that firmware. I was then able to edit the list in a text editor and see that list on the PP. I then updated to 1.6.0 and the list is still there however I can not add to it on the pp again.

gullradriel, 2022-12-30:
I confirm it affects the save load popup. 
I'm checking.

gullradriel, 2022-12-30:
It's not affecting delete / description edition . 
Scanner / Recon are not affected.
I think I'm going to try an update on the freqman save freq func.

grinchdubs, 2023-01-04:
Updated to the new nightly and turns out now I can edit the text file in an editor and still have it work on the PP.

gullradriel, 2023-03-27:
That normal. You have to get down to focus and use rotary encoder to scroll.

closed 2023-03-27 by gullradriel