portapack-mayhem/mayhem-firmware #777: After power cycle all settings are reset to default

Radiomix2000, 2022-12-30:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly verison here: https://github.com/eried/portapack-mayhem/releases)

----

**Describe the bug**
After power cycle all settings are reset to default.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Settings menu'
2. Change something, for example "disable show logo", disable "Hide speaker icon" and just try to set the date/time. 
3. Power off your portapack
4. Switch it back on. You will see, that logo is back, speaker is not shown in notification area and date/time are reset to defaults.

**Expected behavior**
All settings are saved after power cycle.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release: 1.6.0
* Latest Nightly release: -
* Previous working release: 1.3.1

**Additional**
The "save and load app settings" menu don't change the behavior of the bug.
The backup RTC battery on portpack PCB is checked - ok.

iondulgheru, 2023-01-09:
On Portapack H2+ I cannot reproduce this issue on 1.6.0, the settings are saved. I had this problem, but it seems that the backup battery was dead (CR1220). I have a green HackRF clone with Portapack H2+ (the one with Li-Ion battery). I also followed the above steps listed by @Radiomix2000 .

Brumi-2021, 2023-01-09:
Hi @Radiomix2000 , 
I also have the same experience than you using H1 or H2+ . But i can only reproduce  your  problem  when I install any local compile code with my local.   gcc 10.x version (example using  fw 1.6.0 source code).  But the same source code works well when i got the binary compiled with gcc 9.x (example getting it , from here in every official release) .

And I could confirm that this problem started to appear after the merge of that PR #662 (May 28th) . Any local compiled binary prior to that PR#662 works well . You can try 1.5.3 and it should work well , but I guess that you will face that problem from 1.5.4 onwards. Pls try and let us know about it. 

My theory is that,  this PR #662 is using some access to the SD card (to handle the persistent memory) in the limit of the tolerance timings (not robust enough)  , and just adding some different compiled CPU instructions or different HW tolerances (as your unit ) it will not work . I have pending to bisect it and find out which is the critical code part , and try to solve it adding some additional delays or any other workaround . (But not done , because I always try to  work with the latest official  nightly (that works well )

Pls let us know that your set is ok with 1.5.3 or previous , and NG from 1.5.4 onwards .

Brumi-2021, 2023-01-11:
Ok , @Radiomix2000  glad to see that you solved the problem . I was also thinking that other reason could be the cell battery discharge status , but as you indicated that it had a correct voltage I did not mentioned . 
Ok ,finally  bad solder battery holder issue , 
then let's close it 👍

closed 2023-01-11 by Brumi-2021

pinarruiz, 2023-01-12:
Are not app configs stored to the sd card? a battery should not be needed right?

Brumi-2021, 2023-01-12:
Hi @pinarruiz , there are many  data  , like scanner frequency list , captured or Replay  file.C16 data , captured AIS raw frames ,...that are stored in SD card . But in general the user Config App settings,  are stored in "persistent ram memory " that needs  to get also power supply from the  RTC  battery button cell  .

eried, 2023-01-12:
> Are not app configs stored to the sd card? a battery should not be needed right?

I think there was some attempts to move everything (or make it an option) to the sd, but nobody has completed this task

pinarruiz, 2023-01-12:
Thanks @Brumi-2021 and @eried did not read that parto on the wiki, i will include an rtc battery on it