portapack-mayhem/mayhem-firmware #779: Waterfall Color Configuration by File on SD

AlphaGeek1, 2022-12-30:
Request to configure waterfall color palette via file on SD card.
This is related to pull requests #744, #771, and #772, but mostly #744.

Deliver Mayhem with a file on the SD card something like default_palette,txt perhaps in the root directory, along with another directory named something like 'Waterfall_Palettes' and maybe include, original, sunset, and rainbow.

Assuming there is already a routine in the code which opens and reads a text file containing key pairs, this could be used to read in the color palette data with minimal effect on limited memory from storing multiple palettes and additional UI code for selecting them.

AlphaGeek1, 2023-01-19:
@Brumi-2021

Brumi-2021, 2023-01-19:
I th

Brumi-2021, 2023-01-19:
Hi @AlphaGeek1 ,
Probably you are right ,   I think that the best would be to revert that merged commit  and come back to the previous Waterfall LUT colour scheme till someone can introduce a config LUT settings based on  SD files.

Personally I like more the new one  , but I have been trying to capture some AIS signals , and I am not used yet to calibrate by those colours a good S/N capture .  If I adjust the LNA gain to have a clean signal with low background noise, then the replay does not work ... ( the S from S/N looks like to weak) .
Somehow I have not yet calibrated a correct S/N with those colours . Someone already comment that maybe the LUT is too linear colour change and not in dB's , I do not know , but maybe you are right and the easy correction for everybody  would be to come back to the previous one . 

Example those  AIS captured with the new colour scheme , looks clean and good to my eyes,  but when I tried to replay them it  did not work . (they  need to be captured with more GAIN )
[image]
[image]

FYI  those are other stronger  FM and AM signals  comparison : 
[image]
[image]

Cheers,

AlphaGeek1, 2023-01-20:
Agreed.  Regardless of aesthetic preference, you have shown this is a usability issue at the moment.

eried, 2024-01-04:
Just closing this, since it ended being about preferences :)

gullradriel, 2025-04-11:
Done with https://github.com/portapack-mayhem/mayhem-firmware/pull/2617

closed 2025-04-11 by gullradriel