portapack-mayhem/mayhem-firmware #781: i sucseces compile and add test app, but need rx and tx simple code

girs1982, 2023-01-02:
will best if you publish simple code how rx and tx bytes or samples

[image]
[image]

girs1982, 2023-01-22:
how recive samples, simple examle

gullradriel, 2023-03-27:
There are already tons of tutorials everywhere on Google.

closed 2023-03-27 by gullradriel

girs1982, 2023-03-27:
> There are already tons of tutorials everywhere on Google.

where link?

gullradriel, 2023-03-27:
Type "hackrf portapack tutorial" in Google, YouTube, Reddit...

girs1982, 2023-03-27:
> Type "hackrf portapack tutorial" in Google, YouTube, Reddit...

i was try it all, but need sample reciver, how read samples from hackrf in portapack c++ code?

girs1982, 2023-03-27:
> […](#)

no understand code in this , need samle