portapack-mayhem/mayhem-firmware #783: Missing portapack-h1_h2-mayhem.bin in mayhem_v1.6.0_FIRMWARE.zip

KmanOz, 2023-01-05:
This is pretty self explanatory.

mayhem_v1.6.0_FIRMWARE.zip DOES NOT contain portapack-h1_h2-mayhem.bin which is called by the flash_portapack_mayhem.bat file.

I'd say this is going to cause some issues.

Apologies if this has been posted already.

jLynx, 2023-01-05:
Try re download it again. I just checked and it's definitely there
[image]

closed 2023-01-05 by jLynx

KmanOz, 2023-01-05:
Hahahahaha I shit you not. Here's the screen cap. But yes I just downloaded it again and it's there. Weird shit.
[image]

All good thanks for your quick response and great work BTW

jLynx, 2023-01-05:
wow wtf that is odd. Well glad its sorted now :)