portapack-mayhem/mayhem-firmware #788: TPMS transmitter (car tires pressure/temperature monitoring)

simkard69, 2023-01-16:
**Is your feature request related to a problem? Please describe.**
Today HackRF with PortaPack can receive TPMS frames but cannot send them.

**Describe the solution you'd like**
Adding a new transmitter feature for sending TPMS frames over the usual frequencies (315Mhz or 433 Mhz / depending on the country).
This might help the ones (like me) who are trying to diagnose a receiving problem on their car.
When activating new sensors, there is sometimes a need to go and drive for approx. 15 minutes in order to be sure that the car will identify the new sensors ... if car modules configuration is correct.
Having such a feature might avoid driving for nothing just waiting (sometimes in vain) for the car modules to identify the sensors.

**Describe alternatives you've considered**
No other alternatives yet.

**Additional**
Pretty simple, moreover some people already did a solution's concept on that (more like a hack POC) but that might be really cool to have such a feature.

KimIV, 2023-02-06:
It's hardly possible. There are no hardware prerequisites for a TPMS activator.

jLynx, 2023-02-06:
Closed as duplicate of https://github.com/eried/portapack-mayhem/issues/509

closed 2023-02-06 by jLynx

hhjjj75, 2023-02-06:
Look github:txtpms