portapack-mayhem/mayhem-firmware #790: Audio tab not responding

loewal, 2023-01-22:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly verison here: https://github.com/eried/portapack-mayhem/releases)

Flashed 22-01-2023 version

**Describe the bug**

Audio tab does not respond

**To Reproduce**
Steps to reproduce the behavior:
1. Goto Receive
2. Tap on Audio
3. ......

**Expected behavior**
See the receive window.

**Affected versions**
Version 230122 bug!
Version 230106 is working!

Szetya, 2023-01-23:
Delete the SETTINGS folder on the SD card.

loewal, 2023-01-23:
Thnx man,
That did it!

MacGnG, 2023-03-27:
> Delete the SETTINGS folder on the SD card.

This worked for me also! :) THANKS

Brumi-2021, 2023-03-27:
Hello thanks to all , then I think it can be close it .

closed 2023-03-27 by Brumi-2021