portapack-mayhem/mayhem-firmware #791: NRF, BTLE, AFSK crash Mayhem when pressing record button

QwartzNetwork, 2023-01-23:
Hi, Mayhem crash when I press record in some apps like NRF or BTLE

[image]

eried, 2023-01-24:
Is this with an specific version? Have you tried older ones? It might be an issue with your sd or hardware

QwartzNetwork, 2023-01-24:
mayhem 1.6.0, friends have same issue on that version. We think the version is faulty because other recording apps work like a charm

QwartzNetwork, 2023-03-29:
Yes it's automatically recorded in txt's when receving, no needs rec button

gullradriel, 2023-03-31:
There was some history about the same problem in the previously mentioned issue. We'll continue here if anyone is taking on the bug.

gullradriel, 2023-05-10:
After checking it was never implemented. 
Cleaned in https://github.com/eried/portapack-mayhem/pull/968

closed 2023-05-10 by gullradriel