portapack-mayhem/mayhem-firmware #794: world_map.bin missing

TechJeeper, 2023-01-26:
Latest nightly build - world_map.bin is missing from the ADSB directory in the SD card filed.

jLynx, 2023-01-26:
This is a known issue. You will need to get it from the latest stable release. We cannot include it in the nightly builds as it's too big to host on GitHub so our GitHub actions have no where to get it from. Stable builds are diffrent as we manually add it into those builds. Will keep this issue open for others who are curious on why it's missing as well as if anyone has any ideas on how we can host it so it can be downloaded via the actions

TechJeeper, 2023-01-26:
> This is a known issue. You will need to get it from the latest stable release. We cannot include it in the nightly builds as it's too big to host on GitHub so our GitHub actions have no where to get [...]

Thanks!

closed 2023-03-21 by gullradriel